arduino/arduino-ide #396: CTRL + T stops working

tistructor, 2021-06-03:
**Describe the bug**
While using the IDE at some point the CTRL + T auto format command stops working.

**To Reproduce**
Steps to reproduce the behavior:
1. After some time of using the ide, try if auto format still works.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 2.0.0-beta.7
                 Date: 2021-05-17T10:10:22.608Z
                 CLI Version: 0.18.3 alpha [d710b642]

tistructor, 2021-06-10:
In some cases the command is completed in a considerable time, several seconds.

per1234, 2022-06-08:
Thanks so much for your valuable report @tistructor.

I'm happy to tell you that this has been fixed by https://github.com/arduino/arduino-ide/pull/1019

If you would like to give it a try now, it is in the latest nightly build. The download links for the nightly build are listed here:

https://github.com/arduino/arduino-ide#nightly-builds

Otherwise, it will be in the next release of Arduino IDE 2.x

closed 2022-06-08 by per1234