arduino/arduino-ide #397: Fast Serial Port reading makes the IDE cpu usage really high and crashes after some time

alessandromrc, 2021-06-04:
**Describe the bug**
The bug is really annoying because it makes the IDE crash while you're reading a lot of data.

**To Reproduce**
Upload a sketch that prints a lot of data,
Open the serial monitor,
Check the CPU usage of the IDE and after some time it will crash

**Expected behavior**
I expect that the CPU usage won't be really high and that there's something like a garbage collector.

**Screenshots**

[image]

**Desktop**
 - OS: Windows
 - Version: 2.0.0-beta.7

SuperBGLazer, 2021-09-02:
I had the same issue. If I leave the serial monitor running in the background, it sometimes crashes my computer. This issue could be solved by adding an adjustable line limit to the serial monitor.

alessandromrc, 2021-10-30:
Hi @per1234, I can see that the efficiency of the serial monitor is really better but sometimes it loses messages and also it sometimes shows wrong data. I'm using the latest IDE version.

per1234, 2021-10-31:
>I can see that the efficiency of the serial monitor is really better

I'm very glad to hear this!

>but sometimes it loses messages and also it sometimes shows wrong data.

Thanks for reporting that. Please open dedicated issues for these.

There are some known issues with the Serial Monitor, listed here:
https://github.com/arduino/arduino-ide/issues?q=is%3Aissue+is%3Aopen+label%3A%22topic%3A+serial+monitor%22+label%3A%22type%3A+imperfection%22

I didn't spot that matched with your "loses messages". There are a couple that might match to your "sometimes shows wrong data" though, so give that list a quick look in case the issue was already reported.

closed 2021-10-31 by per1234