arduino/arduino-ide #401: "Tools" > "Port" has more options than "Select Other Board & Port"

bb1950328, 2021-06-05:
**Describe the bug**
When I connected a new board to my computer, I first tried to select the correct board and port in the dropdown in the toolbar at the top. I clicked on "Select Other Board & Port". There was a large list of boards. I found the "Arduino Nano" without problems. But on the right side, there's only a message "NO PORTS DISCOVERED". The "Show all ports" checkbox is checked. 
[image]

Then I called `sudo dmesg` to see on which port my Arduino Nano is. Relevant section:
```
[  351.483143] usb 1-2.1.3.3: new full-speed USB device number 15 using xhci_hcd
[  351.584491] usb 1-2.1.3.3: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[  351.584508] usb 1-2.1.3.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[  351.584517] usb 1-2.1.3.3: Product: USB Serial
[  351.653752] usbcore: registered new interface driver usbserial_generic
[  351.653776] usbserial: USB Serial support registered for generic
[  351.655384] usbcore: registered new interface driver ch341
[  351.655408] usbserial: USB Serial support registered for ch341-uart
[  351.655437] ch341 1-2.1.3.3:1.0: ch341-uart converter detected
[  351.656014] usb 1-2.1.3.3: ch341-uart converter now attached to ttyUSB0
```
So my Arduino is on `/dev/ttyUSB0`. Then I discovered "Tools" > "Port" where `/dev/ttyUSB0` is available. 
[image]
After selecting this I was able to upload the blink sketch. (I had to select "Processor" > "Atmega328P (Old Bootloader)", but I dont't think that's part of this problem.)

When I disconnected my Arduino, "Tools" > "Port" > "/dev/ttyUSB0" disappeared so I think the available port detection works there.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect an Arduino
2. Click on Select Other Board & Port
3. The port isn't visible
4. Close the popup
5. Click on "Tools" > "Port"
6. The port is available there

**Expected behavior**
The "Select Other Board & Port" should show the ports from "Tools" > "Port".

**Desktop (please complete the following information):**
 - OS: Ubuntu 21.04
 - Version: `2.0.0 beta7`

**Additional context**
Maybe this is related to #132 but `Master811129` is on Windows while I am on Linux.  
I have a cheap Arduino Nano clone with a CH340G chip.

bugrasan, 2022-03-18:
#681 seems to be related.

ubidefeo, 2022-03-24:
@per1234 
I cannot reproduce this one on Mac OS using RC5 or Nightly

per1234, 2022-03-31:
>https://github.com/arduino/arduino-ide/issues/681 seems to be related.

Thanks for pointing that out @bugrasan!

@bb1950328 does the issue still occur even in the IDE windows that are opened on startup? You can check it by following these instructions:

1. Select **File > Quit** from the Arduino IDE menus if it is running.
1. Start the Arduino IDE.
1. Open the "**Board Selector**" menu.
1. Click on "**Select other board and port...**".
1. Check whether the expected ports are listed under the "**PORTS**" menu.

>I cannot reproduce this one on Mac OS using RC5 or Nightly

@ubidefeo are you able to reproduce it by following the instructions at https://github.com/arduino/arduino-ide/issues/681?

I can still reproduce https://github.com/arduino/arduino-ide/issues/681 with the latest build (2.0.0-rc5-snapshot-c9b498f)

bb1950328, 2022-04-04:
> You can check it by following these instructions:

The `/dev/ttyUSB0` entry appears and disappears in the "PORTS" list when I connect and disconnect the arduino. It has less than one second of delay. The dropdown in the toolbar updates too.  
The "Tools" > "Port" menu did not have the `/dev/ttyUSB0` entry when the arduino was connected on application startup. It appeared when I disconnected it and connected it again.

Tested with `arduino-ide_nightly-20220404_Linux_64bit.AppImage`  
on `Ubuntu 21.10`  
with `Linux bab21desktop 5.13.0-39-generic #44-Ubuntu SMP Thu Mar 24 15:35:05 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`
[stdout (doesn't look interesting to me, but maybe it contains useful information)](https://pastebin.com/Ews5CrXu)

khershberger, 2022-07-11:
I'm having the same issue and discovered an interesting workaround.  I am mystified as to why the workaround works, but I've confirmed that the symbolic link is indeed what makes it work for whatever reason.

If a board is already connected before starting the IDE it will not recognize the board or any serial ports until the board is first disconnected/reconnected.  Workaround is in ~/.arduino15/packages/builtin/tools/serial-discovery/1.3.2 to rename serial-discovery to something else and then create a symbolic link called serial-discovery to the new name.  After doing this the board auto-detection at IDE start correctly detects any boards connected before starting.

Version Info:
arduino-ide_2.0.0-rc8_Linux_64bit.AppImage
NAME="Ubuntu"
VERSION="20.04.4 LTS (Focal Fossa)"

Behavior / Steps to Reproduce:
1.  Connect Arduino Uno to USB port
2. Launch IDE by executing ./arduino-ide_2.0.0-rc8_Linux_64bit.AppImage
3. Click on board/port selector drop down.
4. Observe that the list only contains 'Select other board and port'

Workaournd:
1. In the ~/.arduino15/packages/builtin/tools/serial-discovery/1.3.2 directory:
1.a.   Rename serial-discovery to serial-discovery-bin
1.b    Create symbolic link:      ln -s serial-discovry-bin serial-discovery
  lrwxrwxrwx  serial-discovery -> serial-discovery-bin
   -rwxr-xr-x    serial-discovery-bin
2.  Connect Arduino Uno to USB port
3. Launch IDE by executing ./arduino-ide_2.0.0-rc8_Linux_64bit.AppImage
4. Click on board/port selector drop down.
5. Celebrate that the connected board now appears in the list.

per1234, 2022-09-02:
>The /dev/ttyUSB0 entry appears and disappears in the "PORTS" list when I connect and disconnect the arduino.

OK, I will interpret this information as meaning that the bug is the same as https://github.com/arduino/arduino-ide/issues/681, which has been resolved, so I'll close this as fixed.

>Observe that the list only contains 'Select other board and port'

It sounds like a different bug @khershberger. If it still occurs when using the latest nightly build, please submit a dedicated issue report so that we can effectively track the bug. The nightly build download links are listed here:

https://www.arduino.cc/en/software#nightly-builds

closed 2022-09-02 by per1234