arduino/arduino-ide #402: Command to disconnect the Arduino IDE from the serial port

Lorandil, 2021-06-06:
Often I'm working with more than one open sketch (call them 'One' and 'Two') to check changes
 or adopt/reuse parts of my code.

So I'm working in sketch 'One'.
Than sometimes I want to flash sketch 'Two' just to check something.
Often that's not possibly, because the IDE of 'One' is blocking the serial port (say COM5).
In this situation it would be great if there would be a command to disconect IDE 'One" from COM5.

At the moment I have to close the other IDE (which is really annoying considering the current slow
startup time) to get access to COM5 again.

ubidefeo, 2021-06-06:
hi @Lorandil 
are you using IDE 2.0?

Lorandil, 2021-06-07:
Yes, I'm using the 2.0.0-beta7
Did I miss something ;)?

kittaakos, 2021-06-09:
> At the moment I have to close the other IDE

There is no need to close the other sketch; you can close the serial monitor connection by closing the _Serial Monitor_ view in sketch _"One"_. At least you can avoid one startup time penalty. You can also close the view from the toolbar.

View:
[image]

Toolbar:
[image]

If the serial monitor connection is not closed when you close the _Serial Monitor_ view, it's a bug.

Lorandil, 2021-06-13:
Thanks, that works :)
But as I consider myself an experienced user, I think it would make sense
to add an icon in the serial monitor window to release the port, too.

closed 2021-06-26 by per1234