arduino/arduino-ide #403: Serial Monitor swallows print lines

ullix, 2021-06-07:
On Arduino 2.0 Beta7 the Serial Monitor may swallow `print()` output when followed by 'log_X()' output. To make this visible the Tools->Core Debug Level must be set to at least level X in order to allow log_X() to be printed. This is the simple code used for the printouts:
```
Serial.printf("SerP Wifi begin: %0.1f ms\n", (tb1 - tb0) / 1000.0f);
log_i("logI Total filesystem size: %u kb", SPIFFS.totalBytes() / 1024U);
log_i("logI Used filesystem size:  %u kb", SPIFFS.usedBytes() / 1024U);
log_i("logI System is up.");            
Serial.println("SerP 1 where is my line?");
Serial.printf("SerP 2 Wifi: %0.1f ms\n", (micros() - t0) / 1000.0f);
Serial.printf("SerP 3 Wifi: %0.1f ms\n", (micros() - t0) / 1000.0f);
Serial.printf("SerP 4 Wifi: %0.1f ms\n", (micros() - t0) / 1000.0f);
Serial.printf("SerP 5 Wifi: %0.1f ms\n", (micros() - t0) / 1000.0f);
```
Lines printed with 'Serial.print' contain 'SerP', lines printed with 'log_i()' contain 'logI'. Example output:
```
Arduino 2.0 Beta7
Core Debug Level: Verbose

15:56:04.380 -> SerP Wifi begin: 6.1 ms
15:56:04.380 -> [I][main.ino:217] setup(): logI Total filesystem size: 1345 kb
15:56:04.396 -> [I][main.ino:218] setup(): logI Used filesystem size:  0 kb
15:56:04.397 -> [I][main.ino:219] setup(): logI System is up.
15:56:04.397 -> SerP 3 Wifi: 140.5 ms
15:56:04.397 -> SerP 4 Wifi: 142.4 ms
15:56:04.397 -> SerP 5 Wifi: 144.3 ms
```
Lines printed with log_i() are all printed, but the SerP lines 1 and 2 are missing!

Now when the Tools->Core Debug Level  is set to Error:
```
Arduino 2.0 Beta7
Core Debug Level: Error

16:2:09.417 -> SerP Wifi begin: 1.6 ms
16:2:09.648 -> SerP 1 where is my line?
16:2:09.648 -> SerP 2 Wifi: 120.2 ms
16:2:09.648 -> SerP 3 Wifi: 120.3 ms
16:2:09.648 -> SerP 4 Wifi: 120.3 ms
16:2:09.648 -> SerP 5 Wifi: 120.3 ms
``` 
the log_i() lines are not printed (correct), and the previously missing SerP lines 1 and 2 are shown.

Running the exact same code on Arduino 1.8.13 reveals that the bug is already present in this older version:
```
Arduino 1.8.13
Core Debug Level: Info
16:06:24.791 -> SerP Wifi begin: 6.1 ms
16:06:24.791 -> [I][main.ino:217] setup(): logI Total filesystem size: 1345 kb
16:06:24.824 -> [I][main.ino:218] setup(): logI Used filesystem size:  0 kb
16:06:24.824 -> [I][main.ino:219] setup(): logI System is up.
16:06:24.824 -> SerP 3 Wifi: 142.3 ms
16:06:24.824 -> SerP 4 Wifi: 144.2 ms
16:06:24.824 -> SerP 5 Wifi: 146.2 ms

Arduino 1.8.13
Core Debug Level: Warn
16:08:20.050 -> SerP Wifi begin: 1.6 ms
16:08:20.050 -> SerP 1 where is my line?
16:08:20.050 -> SerP 2 Wifi: 121.6 ms
16:08:20.050 -> SerP 3 Wifi: 121.7 ms
16:08:20.050 -> SerP 4 Wifi: 121.7 ms
16:08:20.050 -> SerP 5 Wifi: 121.7 ms
```
The two SerP 1 and Serp 2 lines are missing from the output when Code Debug Level >= Info is selected.

ullix, 2021-06-12:
Consistent with above observations it seems that output from log_x(), with x=e, w, i, d, v will always print to the monitor, perhaps even while erasing/overwriting/ignoring already printed from Serial.print.

So, if you really want to see any output, print it our via log_x(), and not via Serial!

Don't forget to set the Tools->Core Debug Level to match or you get no output at all! 

If you don't find in the tools menu - I use ESP32 dev module - then restart the Ar IDE (forgetting the menu items seems to be another bug).

per1234, 2021-10-30:
>Running the exact same code on Arduino 1.8.13 reveals that the bug is already present in this older version:

I see. OK, well then I'll close the issue since it is not related to the Arduino IDE code base.

closed 2021-10-30 by per1234