arduino/arduino-ide #405: Compilation speed comparison between stable version and V2 beta under linux

tistructor, 2021-06-09:
I am noticing a consistent speed worsening in the compilation phase between the Arduino IDE versions 1.8.15 and the version 2.0.0.7B under linux Mint 20.1.

With the stable version the compilation speed is significantly faster.
Even when there is no need to recompile the whole project for minimal variation, version 2 beta has disappointing performance.

Watch the two videos:

https://user-images.githubusercontent.com/24445373/121326052-7685b300-c912-11eb-910e-7286bf5d4398.mp4

https://user-images.githubusercontent.com/24445373/121326059-7980a380-c912-11eb-9b79-f0389a462428.mp4

1. Open a project with some complexity
2. Compile the project with the stable version
3. recompile so that most files don't need to be recompiled
4. The speed is very high.

5. Do the same with the Beta version
6. The recompilation takes the same time as the compilation.

I would expect the speeds to be almost identical.

 - OS: Linux Mint 20.1
 - Version: Stable 1.8.15, Beta 2.0.0.7B

Maybe incremental compilation is not enabled?

tistructor, 2021-06-10:
Last night in using it at a certain point the speed increased.
Then it seems that the build log is slower than actually building the project!

per1234, 2021-06-10:
I think you have identified the real problem. I have also noticed that there is a severe bottleneck somewhere in the console output pipeline, as I reported for upload output here: https://github.com/arduino/arduino-ide/issues/68. When there is a lot of output, as there always is when you have verbose output during compilation enabled, the compilation process finishes at the speed dictated by the toolchain as always, but the new IDE must slowly work through displaying the backlog of output.

The effect is likely more noticeable with uploads because you can observe effects of the uploaded sketch starting to run (e.g., LED blinking) while the console shows the upload still happening. So this may be why I only reported the issue as it applies to uploads. I also noticed differences depending on which upload tool was used, so I suppose the same might apply to compilation toolchains.

Lorandil, 2022-01-06:
I have the same issue here, and I wouldn't just call it 'imperfection'.
In my opinion it's really annoying: I have a text segment overflow, the compiler reports an error popup.
The output slows down and goes on for another 10 seconds, before I see how much bytes I'm short...

per1234, 2022-01-06:
Hi @Lorandil. I agree that it is annoying. About the "**type: imperfection**" [label](https://docs.github.com/en/issues/using-labels-and-milestones-to-track-work/managing-labels). This is only a general purpose way to classify issues and PRs are being related to a defect rather than an enhancement. It is not a metric of the severity of the defect. We have additional labels for that purpose, which are applied by the project managers while assigning development priorities:
https://github.com/arduino/arduino-ide/labels?q=criticality

bugrasan, 2022-03-16:
still an issue on linux with `arduino-ide_nightly-20220316_Linux_64bit`

closed 2022-06-14 by davegarthsimpson