arduino/arduino-ide #408: Add native support for Apple Silicon

Logicer16, 2021-06-10:
**Is your feature request related to a problem? Please describe.**
Arduino IDE 2.x runs under Rosetta 2 on Apple Silicon. While this works at the moment, it is more efficient (and will eventually be required) to run a native version.

**Describe the solution you'd like**
Make the macOS version a Universal binary

**Describe alternatives you've considered**
Running under Rosetta 2. *See issues above*

kittaakos, 2021-06-10:
Depends on https://github.com/eclipse-theia/theia/pull/9096.

Edivad99, 2021-08-23:
Now depends on eclipse-theia/theia#9936

LouisCAD, 2022-08-07:
The aforementioned issue has been merged in January this year. What is blocking this now?

Edivad99, 2022-08-08:
https://github.com/arduino/arduino-ide/pull/933

kittaakos, 2022-09-15:
I believe this GH issue is a duplicate of https://github.com/arduino/arduino-ide/pull/933.

I am closing this one. If anyone disagrees, please comment here, and I will reopen if required. Please follow the other issue. Thank you!

per1234, 2022-09-15:
Hi @kittaakos. This is not a duplicate of #933. They are about two similar, but distinct issues:

https://github.com/arduino/arduino-ide/issues/408 is about Arduino distributing builds for Apple Silicon

https://github.com/arduino/arduino-ide/pull/933 is about making it possible for people to build the Arduino IDE on Apple Silicon machines

kittaakos, 2022-09-15:
> #933 is about making it possible for people to build the Arduino IDE on Apple Silicon machines

It's possible to build on M1 once we have the binaries. We have to handle the two things together. Feel free to keep both if managing two tasks is more straightforward than one.

per1234, 2022-09-27:
>We have to handle the two things together.

The ability to build on Apple Silicon machines may be a prerequisite for https://github.com/arduino/arduino-ide/issues/408, but as you explained here there are additional considerations specific to https://github.com/arduino/arduino-ide/issues/408:

https://github.com/arduino/arduino-ide/pull/933#issuecomment-1143221995

>even if we can build the CLI on an M1 chip, it does not fit into the current GitHub Actions-based pipeline: https://github.com/actions/runner-images/issues/2187.

So the two do not need to be handled together. Contributors will benefit from the addition of support for building the Arduino IDE on Apple Silicon machines, so https://github.com/arduino/arduino-ide/pull/933 has merit on its own and should not be blocked by us figuring out the separate logistical issues of setting up our own infrastructure for generating builds for distribution. We will use this issue to track the need for the additional work that is required in order to generate and distribute official builds.

Edivad99, 2022-11-23:
I read this [article](https://github.blog/changelog/2022-08-09-github-actions-self-hosted-runners-now-support-apple-m1-hardware/) and thought that as a solution Arduino could buy a mac mini m1 base and set it as self hosted runners.
I don't know if it can be done or if it has already been rejected as an idea😅

bcr, 2023-07-06:
It has been three years since the transition to Apple Silicon, and no new Macs have an Intel chip. Can I help come up with a roadmap and a set of milestones that will get this transition completed? Most of the IDE seems to run correctly, and this seems to just be the command-line tools causing a problem right now:

```
fork/exec /Users/blake/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags: bad CPU type in executable

➜  ~ file /Users/blake/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags
/Users/blake/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags: Mach-O 64-bit executable x86_64
```

per1234, 2023-07-06:
@bcr this issue is only for the necessary work to the infrastructure of this project to produce native Apple Silicon builds. In fact, we are already distributing such builds but unfortunately we have not yet been able to set up completely automated generation of those builds as is already done with the X86/"Intel" builds.

Requests and work to produce Apple Silicon builds of the various external tool dependencies must be made in the appropriate repository. All the infrastructure is already in place for Arduino CLI and Arduino IDE to use those builds once they are available. In the case of the `builtin:ctags` tool referenced in the error message you shared, the repository is here:

https://github.com/arduino/ctags

For now, please just install Rosetta 2 by following these instructions:

https://support.arduino.cc/hc/en-us/articles/7765785712156-Error-bad-CPU-type-in-executable-on-macOS

After doing that, the error from the ctags command will no longer occur.

Even though it is possible the performance of of an X86 build of **Ctags** running on an Apple Silicon machine via the Rosetta 2 emulation is inferior to what would be achieved with a native Apple Silicon build of **Ctags**, I think we have already achieved the most significant performance improvement by providing the native Apple Silicon build of the Arduino IDE application itself.

closed 2023-10-06 by per1234

jhud, 2025-02-01:
If anyone is looking for a quick fix: 

1. git clone https://github.com/arduino/ctags 
2. remove __attribute__((unused)) from ./general.h file from the line # define __unused__  __attribute__((unused)) and leave # define __unused__
3. ./configure && make
4. copy tags to /Library/Arduino15/packages/builtin/tools/ctags
5. You can now compile.