arduino/arduino-ide #409: Color errors and warnings

tistructor, 2021-06-10:
It would be handy to have the following warnings colored differently: warnings, compilation or linking errors.
For easy identification during the construction process.

Now that the text is monochromatic they are hard to spot.

kittaakos, 2021-06-10:
Possibly the same as https://github.com/arduino/arduino-ide/issues/229. Also related: https://github.com/arduino/arduino-ide/issues/249.

per1234, 2021-06-10:
Thanks @kittaakos! I knew there was one, but couldn't seem to find it earlier.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the https://github.com/arduino/arduino-ide/issues/229.

@tistructor If you end up with additional information to share, feel free to comment in the other thread.

closed 2021-06-10 by per1234