arduino/arduino-ide #410: esptool.py should be ran by python3, not python2

FilipDominec, 2021-06-10:
**Describe the bug**
On newer Ubuntu (Focal Fossa), it is not easy to install the ```pyserial``` module anymore for python2. The repository has pyserial for python3 only. Therefore, this dependency of esptool.py 

**To Reproduce**
Install a recent version of arduino IDE. Connect an ESP32 and try to upload any example. 

**Expected behavior**
Upload should finish successfully, like e.g. on Ubuntu Bionic Beaver.

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version: Focal Fossa, 20.04

**Quick & dirty fix**
Actually, esptool.py seems to run with Python3.

Since the upload command may be hard-coded in Arduino IDE, a small change can be made to switch the python interpreter:

1. go to ```.arduino15/packages/esp32/tools/```  and rename ```esptool.py``` to ```real_esptool.py```
2. in the same directory, create a file ```esptool.py``` with the following contents:

    import subprocess, sys
    subprocess.call("python3 ./real_esptool.py   " + " ".join(sys.argv[1:]), shell=True)

3. run the arduino studio like before; it should be able to upload now

per1234, 2021-06-10:
Hi @FilipDominec. Thanks for your interest in this open source project.

esptool and the choice of which Python interpreter to use is purely a matter of the 3rd party ESP32 boards platform's configuration. The Arduino IDE (or more accurately, Arduino CLI) only provides the framework that runs the compilation and upload processes according to the dictates of the platform configuration of the currently selected board. So this is not anything that can be managed from the code base hosted in this repository. It would be something to be done in the ESP32 boards platform repository, which is not maintained by the Arduino organization:
https://github.com/espressif/arduino-esp32

closed 2021-06-10 by per1234

FilipDominec, 2021-06-10:
@per1234 thanks for info, I reported it here: https://github.com/espressif/arduino-esp32/issues/4717