arduino/arduino-ide #417: Uninstalling a platform or library does not update the version SELECT drop-down

ubidefeo, 2021-06-16:
**Describe the bug**
Installing a platform or a library refreshes the boards/libraries manager tab, displaying the version older than the one just installed in the drop-down menu as a pre-selected option (for downgrade).
Uninstalling leaves the select menu in its last state, with the previous version selected, not the latest available

**To Reproduce**
Steps to reproduce the behavior:
1. Install a platform or a library at its latest version
2. verify that the drop-down changes to the older version
3. uninstall the platform
4. verify that the drop-down has not changed

**Expected behavior**
uninstalling should reset the position of the drop-down

**Screenshots**
before uninstalling
[image]

after uninstalling
[image]

**Desktop (please complete the following information):**
 - OS: Mac OS Big Sur
 - Version: 11.2.3

ubidefeo, 2022-07-29:
this issue is not present in RC9

closed 2022-07-29 by ubidefeo

per1234, 2022-07-30:
I can still reproduce the issue with the latest build (2.0.0-rc9-snapshot-55927ac). However, I think it is fine to consider it a "wontfix", since the behavior is unlikely to cause problems in real world usage.