arduino/arduino-ide #418: Still not compiling with #include <Ethernet.h>

fkromer, 2021-06-17:
**Describe the bug**

I've tried to compile & upload example code of mWebSocketServer to an Arduino Uno + Ethernet Shield v2. The compilation fails.

**To Reproduce**
Steps to reproduce the behavior:
1. optional: Setup Arduino Uno + Ethernet Shield v2
2. Install lib `mWebSockets` into Arduino IDE v2.0.0-beta.5
3. Install lib `Ethernet by Various` like recommended here: https://github.com/arduino/arduino-ide/issues/347#issuecomment-831055324
4. Open example sketch `simple-server`
5. Press button Verify (compilation) or button Upload
6. Error output: 

```
Using board 'arduino_due_x_dbg' from platform in folder: C:\Users\Florian\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12
Using core 'arduino' from platform in folder: C:\Users\Florian\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12
Detecting libraries used...
"C:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "C:\\Users\\Florian\\AppData\\Local\\Temp\\arduino-sketch-3F85081B29165AED3D2476F3E6C74F4E\\sketch\\simple-server.ino.cpp" -o nul
Alternatives for WebSocketServer.h: [mWebSockets@1.4.0]
ResolveLibrary(WebSocketServer.h)
  -> candidates: [mWebSockets@1.4.0]
"C:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "-Ic:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src" "C:\\Users\\Florian\\AppData\\Local\\Temp\\arduino-sketch-3F85081B29165AED3D2476F3E6C74F4E\\sketch\\simple-server.ino.cpp" -o nul
Alternatives for Ethernet.h: [Ethernet@2.0.0 Ethernet@2.0.0]
ResolveLibrary(Ethernet.h)
  -> candidates: [Ethernet@2.0.0 Ethernet@2.0.0]
"C:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "-Ic:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src" "-Ic:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\Ethernet\\src" "C:\\Users\\Florian\\AppData\\Local\\Temp\\arduino-sketch-3F85081B29165AED3D2476F3E6C74F4E\\sketch\\simple-server.ino.cpp" -o nul
Error while detecting libraries included by C:\Users\Florian\AppData\Local\Temp\arduino-sketch-3F85081B29165AED3D2476F3E6C74F4E\sketch\simple-server.ino.cpp
"C:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "-Ic:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src" "-Ic:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\Ethernet\\src" "c:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src\\CryptoLegacy\\BlockCipher.cpp" -o nul
"C:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-IC:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "-Ic:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src" "-Ic:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\Ethernet\\src" "c:\\Users\\Florian\\gitlab\\hurrycane-arduino\\libraries\\mWebSockets\\src\\CryptoLegacy\\Cipher.cpp" -o nul
"C:\\Users\\Florian\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10607 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\\Us [... cut ...]
 Not used: C:\Program Files (x86)\Arduino-1-8-15\libraries\Ethernet
Using library mWebSockets at version 1.4.0 in folder: c:\Users\Florian\gitlab\hurrycane-arduino\libraries\mWebSockets 
Using library Ethernet at version 2.0.0 in folder: c:\Users\Florian\gitlab\hurrycane-arduino\libraries\Ethernet 
Compilation error: Error: 2 UNKNOWN: exit status 1
```

**Expected behavior**
The example sketch compiles.

**Screenshots**

n.a.

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Version: 2.0.0-beta.5

**Additional context**

n.a.

fkromer, 2021-06-17:
I've tried to do a clean uninstall of all Arduino IDEs (v1.8.5 + v2.0.0-beta.5) on the machine. I had to uninstall a few directories in `C:\Users\<USER>\Documents\` manually: `.arduinoIDE`, `Arduino`, `ArduinoData`. I'm not sure if I've deleted everything cause there is no documentation about the location the IDE saves data into. I reinstalled the Arduino IDE from the Windows 10 store (I've no idea if this makes a difference in comparison to installing from installer), reinstalled the board support for Arduino Uno (like described here: https://docs.arduino.cc/software/ide-v1/tutorials/getting-started/cores/arduino-sam) and the lib `mWebSockets` again. Still, if I try to compile I get an error message containing a false positive (the lib `mWebSockets` does support the Arduino Due for sure):

```
In file included from C:\Users\Florian\Documents\Arduino\libraries\mWebSockets\src\base64\Base64.cpp:2:0:
C:\Users\Florian\Documents\Arduino\libraries\mWebSockets\src\base64\../platform.h:30:4: error: #error "Unsupported platform"
 #  error "Unsupported platform"
    ^
C:\Users\Florian\Documents\Arduino\libraries\mWebSockets\src\base64\Base64.cpp:8:24: fatal error: pgmspace.h: No such file or directory
 #  include <pgmspace.h>
                        ^
compilation terminated.
exit status 1
Error compiling for board Arduino Due (Programming Port).
```

per1234, 2021-06-17:
Hi @fkromer Thanks for your interest in this open source project.

As the error message says, the library does not support the Arduino Due board you are compiling for:
https://github.com/skaarj1989/mWebSockets/blob/1.4.0/src/platform.h#L19-L31

I see you mention the Arduino Uno several times. Yet you are compiling for the Arduino Due. If you have an Arduino Uno then you must install "Arduino AVR Boards" via the Boards Manager and then select **Tools > Board > Arduino AVR Boards > Arduino Uno**.

If you need further assistance, feel free to post on the Arduino Forum. We'll be happy to help you out over there:
https://forum.arduino.cc/

closed 2021-06-17 by per1234

fkromer, 2021-06-17:
@per1234 Sorry, that was a typo. It's about the Due. The issue is still unsolved. However I decided to stick to Arduino IDE v.1.8.* for now. After a cleaning up the filesystem from old files this IDE version was able to compile and upload.