arduino/arduino-ide #419: Provide an option for a complete recompile, including any board specific cached code

ullix, 2021-06-18:
The pre-compiling and caching is really convenient most of the time, but sometimes I make changes to e.g. the board specific code and want the IDE to compile everything again, including all libs, and all core, simply everything.

I suggest to make the 'Verify'  and 'Upload' buttons respond with a complete recompile when `CTRL` or `CTRL SHIFT` is pressed during clicking.

per1234, 2021-06-19:
Hi @ullix. Would you mind providing a specific and reproducible example of when you found a recompile to be necessary? You mentioned something in[ our previous discussion on the forum](https://forum.arduino.cc/t/can-i-force-the-arduino-ide-to-recompile-everything/866547/5?u=pert), but there was not enough information to allow me to investigate.

The only problem I have run into with the caching is that compiler warnings for the cached code will only be shown on the first compilation, as reported and discussed here: https://github.com/arduino/arduino-cli/issues/1008

My opinion is that the Arduino build system should be able to always do the right thing with caching automagically. This avoids the need for increasing the complexity of the Arduino IDE's UI to add a capability only 0.1% of users have any use for. There is a proposal in https://github.com/arduino/arduino-cli/issues/1008 for just such an automagic solution for the disappearing warnings issue.

ullix, 2021-06-19:
> ... providing a specific and reproducible example of when you found a recompile to be necessary

Sure, I have an ongoing, very major problem with code for an ESP32. In short: it is impossible to reliably download large files (>150k) https://github.com/me-no-dev/ESPAsyncWebServer/issues/984

In investigating this it was necessary to add code to some libs (ESPAsyncWebServer, AsyncTCP, WiFi, other), which required a recompile of those libs. Furthermore, some testing also required to modify code of the IDF components. So, even the full core had to be recompiled.

The Ar IDE 2.0 promises to be a lot more attractive even for people, who are now moving to more professionally platforms, so the share of people wanting currently little used feature will increase. The corollary is, if you explicitly do not cater for those people, you will not get them, and will maintain and even underline Ar IDE as a beginners platform.

Sometimes things do go wrong in unexpected ways. And you will surely reach the point - correctly or incorrectly - to suspect that something in the cached code is wrong. As you had said, in the old IDE you had the feature that the old cache was deleted upon exiting the IDE, however, this option no longer exists. So, if you want to re-establish the cache you have to resort to some odd work-arounds. I now use (on Ubuntu Linux) the deletion of all things `arduino-*` in `/tmp`.

As I suggested, the buttons/functions 'compile' and 'upload'  could be used with an additional `CTRL` or `CTRL-SHIFT`. Perhaps even `CTRL` for recompiling all libs except the core, and `CTRL-SHIFT` by also recompiling the core. 

This leaves the choice to the user. I can't see that such a feature would blow up the code.

per1234, 2021-06-19:
Core and library files are recompiled if they have been modified since they were cached:
https://github.com/arduino/arduino-cli/blob/0.18.3/legacy/builder/builder_utils/utils.go

As long as that is working correctly, it will never be necessary for the user to take some manual action via the IDE's GUI to trigger a complete recompile because they already did that when they modified the file. It just does the right thing automatically.

If the system for determining when to recompile is not working correctly (it works just as expected during my quick tests), then we should fix the system rather than adding UI as a bandaid. So if you have found some conditions where a core or library should have been recompiled and it was not, then please submit a bug report to https://github.com/arduino/arduino-cli/issues with detailed step by step instructions we can easily follow to reproduce the issue.

ullix, 2021-06-20:
> It just does the right thing automatically.

I tested this on a few things of interest to me, and found that indeed a re-compile was forced. And, even more impressively, when a change was made at a point, which necessitated the re-compile on other files, in which I had NOT made a change, apparently those were also made. Example:

On an ESP32 the download of large files still consistently results in WDT induced crash, with the task "async_tcp" being the culprit. This task is defined in AsyncTCP.cpp. The header file AsyncTCP.h has this definition:

```
//If core is not defined, then we are running in Arduino or PIO
#ifndef CONFIG_ASYNC_TCP_RUNNING_CORE
#define CONFIG_ASYNC_TCP_RUNNING_CORE -1 //any available core
#define CONFIG_ASYNC_TCP_USE_WDT 1 //if enabled, adds between 33us and 200us per event
#endif
```
The hope was that by fixing the core for this task to core 1, the WDT limitation could be overcome (did not work). Of interest here is that this change in the header file alone resulted in this excerpt from compilation results:

```
... bin/xtensa-esp32-elf-g++ ... ESPAsyncWebServer-master/src/AsyncEventSource.cpp -o /dev/null
... bin/xtensa-esp32-elf-g++ ... ESPAsyncWebServer-master/src/AsyncWebSocket.cpp -o /dev/null
Using cached library dependencies for file: ESPAsyncWebServer-master/src/WebAuthentication.cpp
... bin/xtensa-esp32-elf-g++ ... ESPAsyncWebServer-master/src/WebHandlers.cpp -o /dev/null
... and more
```
So, some files of the ESPAsyncWebServer had to be recompiled, and some not. While I find this auto-detection quite impressive, I also find that I would never 100% trust the validity of such auto-detection under all circumstances!  Here I will certainly always force a full recompilation.

closed 2021-06-21 by per1234

php4fan, 2023-12-06:
> Core and library files are recompiled if they have been modified since they were cached:
> As long as that is working correctly, 

It doesn't.
And regardless of whether or not you fix that, just provide a way to clear the cache / force everything to be recompiled. The inability to do so is simply ridiculous.

php4fan, 2023-12-06:
BTW what is the workaround? I tried deleting /tmp/arduino-* and that didn't work. Where are the precompiled files stored?