arduino/arduino-ide #421: Compilation error: Error: 2 UNKNOWN: no FQBN provided while board is in and plugged

jerabaul29, 2021-06-18:
This may be a duplicate of #151 in which case feel free to close. But it may be another issue (see the "greyed board problem lower down).

- host OS: Ubuntu 20.04

- ide version etc: installed today from your pages:

```
Version: 2.0.0-beta.7
Date: 2021-05-17T10:10:01.588Z
CLI Version: 0.18.3 alpha [d710b642]
```

# steps

- I installed a board (in my case, the sparkfun apollo3 boards, by adding their board manager url to the preferences URLs list, and then installing from the boards manager). The boards seem to be well present and installed, see the screenshot:

[image]

- I plugged in the board inside my computer, it is recognized and shows up on the ttyUSB0; I choose the right board and port. **One thing seems strange there though, the board name is greyed out.** See screenshot:

[image]

- From now on, I get the "message 2 FQBN" issue, see screenshot.

[image]

Any idea what causes that?

per1234, 2021-06-19:
Hi @jerabaul29. Do you see the boards listed under the Arduino IDE's **Tools > Board > Sparkfun Apollo3 Boards** menu?

jerabaul29, 2021-06-19:
No, I do not see the boards there, see screenshot:

[image]

However in the board manager these boards appear as installed:

[image]

per1234, 2021-06-19:
Please post the output you get in the Linux terminal after starting the Arduino IDE.

jerabaul29, 2021-06-19:
Here it is :) 

```
Starting backend process. PID: 8416
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root WARN please install @theia/electron@1.12.0-next.1a1d7afc as a runtime dependency
root INFO Configuration directory URI: 'file:///home/jj/.theia'
root INFO Theia app listening on http://localhost:42767.
root WARN The local plugin referenced by local-dir:/home/jj/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/home/jj/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:/home/jj/.arduinoProIDE/plugins does not exist.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.1-beta.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO Deploying backend plugin "cortex-debug@0.3.10" from "/home/jj/Desktop/Jean/Software/arduino/arduino-ide_2.0.0-beta.7_Linux_64bit/resources/app/plugins/cortex-debug/extension/dist/extension"
root INFO Deploying backend plugin "vscode-arduino-tools@0.0.1-beta.1" from "/home/jj/Desktop/Jean/Software/arduino/arduino-ide_2.0.0-beta.7_Linux_64bit/resources/app/plugins/vscode-arduino-tools/extension/dist/bundle"
root INFO Deploying backend plugin "cpp@1.52.1" from "/home/jj/Desktop/Jean/Software/arduino/arduino-ide_2.0.0-beta.7_Linux_64bit/resources/app/plugins/vscode-builtin-cpp/extension"
root INFO Deploying backend plugin "json@1.46.1" from "/home/jj/Desktop/Jean/Software/arduino/arduino-ide_2.0.0-beta.7_Linux_64bit/resources/app/plugins/vscode-builtin-json/extension"
root WARN Backend ConfigServiceImpl.onStart is slow, took: 54.1 ms
daemon INFO Starting daemon from /home/jj/Desktop/Jean/Software/arduino/arduino-ide_2.0.0-beta.7_Linux_64bit/resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
root INFO Deploying backend plugin "json-language-features@1.46.1" from "/home/jj/Desktop/Jean/Software/arduino/arduino-ide_2.0.0-beta.7_Linux_64bit/resources/app/plugins/vscode-builtin-json-language-features/extension/client/dist/jsonMain"
root INFO Deploy plugins list took: 56.0 ms
daemon INFO Using config file: /home/jj/.arduinoIDE/arduino-cli.yaml
daemon INFO arduino-cli version 0.18.3
daemon INFO Setting up Prometheus metrics on :9090/metrics
daemon INFO Starting daemon on TCP address 127.0.0.1:50051
daemon INFO Daemon is now listening on 127.0.0.1:50051...
daemon INFO Daemon is running.
daemon INFO Checking signature [index: /home/jj/.arduino15/package_index.json, signaturefile: /home/jj/.arduino15/package_index.json.sig, trusted: true]
daemon INFO Checking signature [error: opening signature file: open /home/jj/.arduino15/package_sparkfun_apollo3_index.json.sig: no such file or directory, index: /home/jj/.arduino15/package_sparkfun_apollo3_index.json, signaturefile: /home/jj/.arduino15/package_sparkfun_apollo3_index.json.sig]
daemon INFO Loading hardware from: /home/jj/.arduino15/packages
daemon INFO Loading package SparkFun from: /home/jj/.arduino15/packages/SparkFun/hardware
daemon INFO Checking signature [error: opening signature file: open /home/jj/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/installed.json.sig: no such file or directory, index: /home/jj/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/installed.json, signaturefile: /home/jj/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/installed.json.sig]
daemon INFO Loaded platform [platform: {ID:"SparkFun:apollo3",Installed:"1.2.2",Latest:"2.1.0",Name:"SparkFun Apollo3 Boards"}]
daemon INFO Checking existence of 'tools' path: /home/jj/.arduino15/packages/SparkFun/tools
daemon INFO Loading tools from dir: /home/jj/.arduino15/packages/SparkFun/tools
daemon INFO Loaded tool [tool: Version: 8-2018-q4-major Systems: [OS: x86_64-apple-darwin, OS: i386-apple-darwin11, OS: i686-linux-gnu, OS: x86_64-pc-linux-gnu, OS: i686-mingw32]]
daemon INFO Loading package builtin from: /home/jj/.arduino15/packages/builtin
daemon INFO Checking existence of 'tools' path: /home/jj/.arduino15/packages/builtin/tools
daemon INFO Loading tools from dir: /home/jj/.arduino15/packages/builtin/tools
daemon INFO Loaded tool [tool: Version: 5.8-arduino11]
daemon INFO Loaded tool [tool: Version: 1.2.1]
daemon INFO Adding libraries dir [dir: /home/jj/Arduino/libraries, location: user]
daemon INFO Adding libraries dir [dir: /home/jj/.arduino15/packages/SparkFun/hardware/apollo3/1.2.2/libraries, location: platform]
root INFO Git askpass helper is listening on http://127.0.0.1:43633.
root INFO Using Git [2.25.1] from the PATH. (/usr/bin/git)
root WARN Frontend CommonFrontendContribution.configure is slow, took: 129.9 ms
root WARN Failed to resolve module: @types/body-parser
root WARN Failed to resolve module: @types/cookie
root WARN Failed to resolve module: @types/express
[... cut ...]
root INFO [84fa8903-7c13-42c6-a7ec-79d623fe47b0][marus25.cortex-debug]: Started plugin.
root INFO [84fa8903-7c13-42c6-a7ec-79d623fe47b0][arduino.vscode-arduino-tools]: Started plugin.
root INFO [84fa8903-7c13-42c6-a7ec-79d623fe47b0][vscode.cpp]: Started plugin.
root INFO [84fa8903-7c13-42c6-a7ec-79d623fe47b0][vscode.json]: Started plugin.
root INFO [84fa8903-7c13-42c6-a7ec-79d623fe47b0][vscode.json-language-features]: Started plugin.
root INFO [84fa8903-7c13-42c6-a7ec-79d623fe47b0] Start of 5 plugins took: 2311.2 ms
```

jerabaul29, 2021-06-19:
(just launching as ```$ ./arduino-ide```.

SamGeens, 2021-07-06:
I updated from beta.5 to beta.8 today and am now experiencing the same problem (Windows). The Generic ESP boards are now grayed out and the same message occurs `Compilation error: Error: 2 UNKNOWN: no FQBN provided`.
[image]