arduino/arduino-ide #427: ESP8266/FTDI Serial Console Issue

nboorstin, 2021-06-28:
**Describe the bug**
Hi, I'm using an ESP8266 connecting to USB via a FTDI FT231XS using the ESP8266 Boards version 2.7.4 library (https://github.com/esp8266/Arduino), and although I can compile and upload properly using the IDE 2.0.0-beta.7, I can't see any output from the Serial Monitor: I only get this symbol: �. The Serial Monitor works as expected on the old Arduino IDE.

**To Reproduce**
Steps to reproduce the behavior:
Plug in the board, select the port and board type, try to open the Serial Monitor

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 2.0.0-beta.7

per1234, 2021-06-30:
Hi @nboorstin. Do you have the baud rate menu in the Arduino IDE 2.x Serial Monitor set to match the rate set in the `Serial.begin()` call in your sketch?

For example, if your sketch contains this line:
```cpp
Serial.begin(115200);
```
Then you must make sure that the Serial Monitor's dropdown menu is set to "115200 baud". Mismatched baud rates is the most common cause of getting gibberish characters in Serial Monitor.

nboorstin, 2021-06-30:
Hi @per1234, thanks for taking the time to respond!
Both baud rates are set to 9600, but I only get � as output. (I get � in the IDE 2.x; in the old IDE I get the proper serial output). 
Thanks!

per1234, 2021-06-30:
Do you get the same problem if you uploaod this super simple sketch?
```cpp
void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println("hello");
  delay(1000);
}
````

nboorstin, 2021-07-01:
Yes, same problem, it compiles/uploads with both the old and new IDE but the serial monitor only works in the old IDE

JonathanKartun, 2021-07-22:
I found something else with serial monitor.

If it logs to Serial Monitor.
Firstly I'm using mac. If I press Mac+A to select all. It selects my code.
Secondly, if I manually select the logged text to copy. It copies with an upside down question mark .
Also it selects the text all the way up to the code - including the UI Elements and could even select the line numbers.

per1234, 2021-12-31:
>If I press Mac+A to select all. It selects my code.

I think we have been using https://github.com/arduino/arduino-ide/issues/105 to track the lack of correct Select All capability: 
>We get copy/paste/find/select-all for the Serial Monitor for free.

but https://github.com/arduino/arduino-ide/issues/105 is more of a feature request/enhancement proposal than a defect report so you are welcome to open a dedicated issue for it,

> if I manually select the logged text to copy. It copies with an upside down question mark on each line.

Maybe it is the same as this?: https://github.com/arduino/arduino-ide/issues/730

I don't think either of these issues you observed are directly related to the corruption that is the subject of this issue.