arduino/arduino-ide #430: Set standard c++ version to 17 instead of 11

FredM67, 2021-06-30:
In the file "platform.txt", replace "-std=gnu++11" with "-std=gnu++17" to support the latest features of C++.

per1234, 2021-06-30:
Hi @FredM67. Thanks for your interest in this open source project.

This repository contains the code for the Arduino IDE. There is no platform.txt here, only the code for a GUI. The author of each boards platform can chose to use any compiler compiler flags they like, so the Arduino IDE has no influence over which C++ standard is used for a given board.

closed 2021-06-30 by per1234

Silver-Fang, 2021-11-14:
> Hi @FredM67. Thanks for your interest in this open source project.
>
> This repository contains the code for the Arduino IDE. There is no platform.txt here, only the code for a GUI. The author of each [...]

So which repository contains that?

per1234, 2021-11-14:
>which repository contains that?

There are multiple repositories. It depends on which board you are using. You can see a fairly comprehensive list here:
https://github.com/per1234/ino-hardware-package-list/blob/master/ino-hardware-package-list.tsv

FredM67, 2021-11-14:
That should be at least this one:
https://github.com/arduino/ArduinoCore-avr