arduino/arduino-ide #431: indexing operation never completes

danyhm, 2021-07-01:
**Describe the bug**
after opening some sketches , sometimes the indexing operation at the bottom of the IDE is stuck on a number and never completes.

**To Reproduce**
randomly appears. in my case in esp32 examples.

**Expected behavior**
the operation should complete in a timely manner.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 2.0.0 beta 7

**Additional context**

ubidefeo, 2022-01-25:
I am no longer able to reproduce this bug

closed 2022-01-25 by fstasi

per1234, 2022-01-25:
I suspect there have been multiple causes of this symptom over time. One specific cause was fixed by https://github.com/arduino/arduino-language-server/pull/96 and I would guess the other work on the Arduino Language Server fixed some others.

I have not been able to reproduce it ever since I started using [Arduino Language Server 0.6.0](https://github.com/arduino/arduino-language-server/releases/tag/0.6.0), which was added to the Arduino IDE build by https://github.com/arduino/arduino-ide/pull/738.