arduino/arduino-ide #432: Library SD missing, had to add manually.

brunotacca, 2021-07-02:
**Describe the bug**
IDE doesn't install or can't find SD library. I had to download and add it manually at Arduino/libraries. Boards installed:

- Arduino SAMD Boards (32-bits ARM Cortex-M0+) by Arduino (latest)
- TinyCircuits SAMD Boards by TinyCircuits (latest - [json](http://files.tinycircuits.com/ArduinoBoards/package_tinycircuits_index.json)) (not needed to reprocude)

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh install 2.0.0-beta.7
2. Download the boards mentioned above. 
3. Example blink
4. Select any board
5. Include libraries `#include <SPI.h>` `#include <SD.h>`
6. Verify

**Expected behavior**
It should find the SD library as soon as a board is selected. Since it's mentioned [here](https://www.arduino.cc/reference/en/libraries/sd/) that it's compatible with all boards.

**Screenshots**
[image]

**Desktop (please complete the following information):**
- Ubuntu 20.04.1 LTS
- Kernel 5.8.0-59-generic
- AMD Ryzen 7 2700X

**Additional context**

The output I got is that it can't find the SD.h at the places it's looking for.

```
Using board 'tinyscreen' from platform in folder: /home/tacca/.arduino15/packages/TinyCircuits/hardware/samd/1.1.0
Using core 'arduino' from platform in folder: /home/tacca/.arduino15/packages/TinyCircuits/hardware/samd/1.1.0
Detecting libraries used...
...
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
...
Alternatives for STBLE.h: [STBLE@0.1.0]
ResolveLibrary(STBLE.h)
  -> candidates: [STBLE@0.1.0]
/home/tacca/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x03EB -DUSB_PID=0x8009 -DUSBCON "-DUSB_MANUFACTURER=\"TinyCircuits\"" "-DUSB_PRODUCT=\"TinyScreen+\"" -DUSBCON -I/home/tacca/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/home/tacca/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/ -I/home/tacca/.arduino15/packages/TinyCircuits/hardware/samd/1.1.0/cores/arduino -I/home/tacca/.arduino15/packages/TinyCircuits/hardware/samd/1.1.0/variants/tinyscreen_p -I/home/tacca/.arduino15/packages/TinyCircuits/hardware/samd/1.1.0/libraries/SPI -I/home/tacca/Arduino/libraries/STBLE/src /tmp/arduino-sketch-2D2A844382121D13431A39C08EFA2ECB/sketch/sensor.ino.cpp -o /dev/null
Alternatives for SD.h: []
ResolveLibrary(SD.h)
  -> candidates: []
Using library SPI at version 1.0 in folder: /home/tacca/.arduino15/packages/TinyCircuits/hardware/samd/1.1.0/libraries/SPI 
Using library STBLE at version 0.1.0 in folder: /home/tacca/Arduino/libraries/STBLE 
Compilation error: Error: 2 UNKNOWN: exit status 1
```

I might be missing something really dumb since I couldn't find anything related, nonetheless thank you all in advance.

per1234, 2021-07-02:
Hi @brunotacca. Thanks for taking the time to submit a bug report.

From classic Arduino IDE, we are accustomed to having a collection of common libraries pre-installed, which includes the SD library.  However, Arduino IDE 2.x does not currently come with any libraries pre-installed. So you must install any library your sketch uses. So the error you got is normal and expected.

The Library Manager (**Sketch > Include library > Manage Libraries** is a convenient way to install any libraries you need.

closed 2021-07-02 by per1234

brunotacca, 2021-07-02:
@per1234 Hello, thank you for your quick response.

I actually have tried to add SD library by the library manager and wasn't able to find it.
Thus, now that you mentioned I went a little deep investigating the libraries showing there and found out what is the problem.

Maybe the Library Manage as a fixed size for results.
When typing "SD" at the search bar, it doesn't show the "SD by Arduino, SparkFun" library.
However, if I increase the terms, like typing "SD Arduino" it appears near the end of the list.
If I may, I would suggest enhancing this feature by making it load more results as the scroll reaches the end.

Nonetheless, thank you very much for your help.