arduino/arduino-ide #433: ctags: bad CPU type in executable in macOS (2.0.0-beta.7)

hgrecco, 2021-07-03:
**Describe the bug**
Cannot Verify/Compile even simple examples (AnalogReadSerial.ino). ctags: bad CPU type in executable in macOS with

**To Reproduce**

Just run the any example.

> 
> Using board 'uno' from platform in folder: /Users/grecco/Library/Arduino15/packages/arduino/hardware/avr/1.8.3
> Using core 'arduino' from platform in folder: /Users/grecco/Library/Arduino15/packages/arduino/hardware/avr/1.8.3
> Detecting libraries used...
> /Users/grecco/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/grecco/Library/Arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/Users/grecco/Library/Arduino15/packages/arduino/hardware/avr/1.8.3/variants/standard /var/folders/xj/xsrq0ml15_q98sq78466mbwr000bjr/T/arduino-sketch-ECD2946323C50FF71A5DD32C03F77EE8/sketch/AnalogReadSerial.ino.cpp -o /dev/null
> Generating function prototypes...
> /Users/grecco/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Users/grecco/Library/Arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/Users/grecco/Library/Arduino15/packages/arduino/hardware/avr/1.8.3/variants/standard /var/folders/xj/xsrq0ml15_q98sq78466mbwr000bjr/T/arduino-sketch-ECD2946323C50FF71A5DD32C03F77EE8/sketch/AnalogReadSerial.ino.cpp -o /var/folders/xj/xsrq0ml15_q98sq78466mbwr000bjr/T/arduino-sketch-ECD2946323C50FF71A5DD32C03F77EE8/preproc/ctags_target_for_gcc_minus_e.cpp
> /Users/grecco/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /var/folders/xj/xsrq0ml15_q98sq78466mbwr000bjr/T/arduino-sketch-ECD2946323C50FF71A5DD32C03F77EE8/preproc/ctags_target_for_gcc_minus_e.cpp
> Compilation error: Error: 2 UNKNOWN: fork/exec /Users/grecco/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags: bad CPU type in executable

**Expected behavior**
It should compile (and it does with Arudino 1.8.13.

**Desktop (please complete the following information):**
 - OS: macOS Big Sur (11.4). MacBook Pro 2017
 - IDE: Arduino IDE 2.0.0-beta.7

**Additional context**
I also have Arduino 1.8.13 installed and there it works fine. It seems to be using a differen toolchain located in: 

```
/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags
```

flatsiedatsie, 2021-12-24:
I'm seeing this issue on M1 Macbook Pro, running Monterey 12.1. 
Upgrading to latest Arduino IDE 1.8.19 did not solve it.

It happened while trying to upload code to an Arduino Nano
There is no problem with the checkmark (test) button, only when using the arrow (upload) button.
I tried updating all libraries too, no effect.
This also occurs with the basic blink sketch.

Could it be that part of the upload process is still 32 bit?

nickdesaulniers, 2023-01-15:
possible dup of #1818 or #1796.  @hgrecco @flatsiedatsie do you have rosetta installed?

flatsiedatsie, 2023-01-16:
@nickdesaulniers Yes I have

per1234, 2025-01-08:
Thanks for your report @hgrecco. Although Arduino IDE 2.x is a native Apple Silicon application, it uses various helper tools and some of these tools are only available in "Intel" (x86) builds. One such tool is **ctags**.

Apple has created a "binary translator" named **Rosetta 2** that allows x86 applications to run on Apple Silicon machines. So the problem can be solved by simply installing **Rosetta 2**. You can do that by following these instructions:

https://support.apple.com/102527

We are tracking the need to produce native Apple Silicon builds of **ctags** to avoid the need for **Rosetta 2** here: https://github.com/arduino/ctags/issues/20. The infrastructure is already present in Arduino IDE to automatically install the Apple Silicon build of the the helper tools if available, so no work is needed in this repository in order to resolve this deficiency. For this reason, and since the deficiency is already being tracked in the relevant repository, I'll go ahead and close this issue.

closed 2025-01-08 by per1234

ccashman, 2025-04-20:
> So the problem can be solved by simply installing **Rosetta 2**.

The problem of running the Arduino IDE 2.x can be _worked around_ by installing Rosetta 2. The actual problem described here — ctags not natively compiling for Apple Silicon — cannot be worked around by installing Rosetta.

Even for running Arduino IDE 2.x, there’s no assurance that Rosetta is a long-term workaround. Apple has not committed to Rosetta as a permanent feature of macOS, and if we look to the original version, Rosetta 1 only had a lifespan of about five years (2006-2011) before it was removed from the OS.