arduino/arduino-ide #434: Preference getting restored every time when i open IDE or select a new board option in window 10

forvishnu, 2021-07-04:
Preferences option getting restored to default settings when i open/restart IDE or select a new board. Autocomplete function is disabled automatically every time. board selection too gets restored sometimes.

per1234, 2021-07-05:
There are several other reports about specific symptoms of this bug with the preferences not being preserved:

- https://github.com/arduino/arduino-ide/issues/313
- https://github.com/arduino/arduino-ide/issues/315
- https://github.com/arduino/arduino-ide/issues/331

I have been frustrated by it myself.

per1234, 2021-11-27:
Hi @forvishnu. We are still waiting for you to provide the information requested in the previous reply so that we can proceed with the investigation or evaluation of this.

per1234, 2022-01-10:
Closing due to lack of response.

@forvishnu if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

closed 2022-01-10 by per1234