arduino/arduino-ide #435: Support for Cortex-Debug serverArgs and JLink script files

Aurical, 2021-07-05:
**Is your feature request related to a problem? Please describe.**
I would like to ask is there is a way to pass in Arduino platform.txt relative directories into the Cortex-Debug launch.json serverArgs field. If not, consider this a request for the feature.

When trying to add in debugging support for a 3rd party board, I find that I need to give the JLink gdb server a JLinkscript file for debugging to work properly. This can be achieved by including 
 `"serverArgs": ["-scriptfile", "{pathToFile}\\script.JLinkScript"]` in the debug_custom.json file in the sketch directory, which will be included into the launch.json Cortex-Debug config file. 
Ideally, the JLinkscript file would be included in the tools or variant directory of the Arduino package for the board, but I have been unable to find a way to pass in the relative file directory from platform.txt into the `"serverArgs": [ ]` field. Using `debug.server.jlink.script={runtime.tools.path}/debug/script.JLinkScript` just puts it in the `"configFiles": [ ]` field instead, which does not work with JLink gdb server.

**Describe the solution you'd like**
It would be best if `debug.server.jlink.script={}` can directly cause the IDE to add in `"-scriptfile" "{pathToFile}/script.JLinkScript"` when invoking JLinkGDBserverCL.exe, with the ability to use platform.txt file path variables for the script file.
An alternative would be a way to add in `"serverArgs": [ ]` through platform.txt.

ubidefeo, 2021-07-07:
@Aurical have you tried overriding the automatically generated properties?
We have yet to document this because we'll rework how platforms enable debugging, so it's still very experimental and we officially only enabled it for SAMD.
Essentially you can place a file `debug_custom.json` in your sketch and it will override any property and add the ones you decide to implement yourself.

As an example check this comment I posted in an old issue
https://github.com/arduino/arduino-ide/issues/87#issuecomment-774362701

Aurical, 2021-07-08:
@ubidefeo Yes I have and it works as expected. 
However, there is no easy way for the `debug_custom.json` file to get the Arduino15 folder path directly, thus requiring manual input from the user, and not all users are familiar with where downloaded board packages are located for their system.
It is good to hear that debugging will be reworked, and I hope that script file support for JLinks can be improved, since it is one of the more commonly used debug adaptor hardware.