arduino/arduino-ide #436: Scroll bar width non-standard

tomotvos, 2021-07-06:
**Describe the bug**
The width of the vertical scroll bar in sketches is too narrow on IDE 2

**To Reproduce**
Steps to reproduce the behavior:
1. Edit a sketch with more than a full screen of lines
2. Scroll bar appears roughly half the width it should be

**Expected behavior**
The scroll bar should be the standard size for each platform

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: Big Sur

**Additional context**
none

per1234, 2022-11-07:
The Theia developers set this non-standard width without providing any justification for doing so:

https://github.com/eclipse-theia/theia/pull/852

The scrollbar width is intended to be configurable, but unfortunately the non-standard width was configured in such a way that causes that configuration to be ignored:

https://github.com/eclipse-theia/theia/issues/11732

My hope is that the bug can be resolved in Theia. If that ends up not being possible, I think it will be important for Arduino to investigate whether there is a way to patch it from the Arduino IDE codebase and ideally to also set the default width in Arduino IDE to the same width as the scrollbar in every non-Theia application ever made.

per1234, 2023-08-01:
The fix for https://github.com/eclipse-theia/theia/issues/11732 was released in Theia 1.33.0. It was introduced into this project via https://github.com/arduino/arduino-ide/pull/2027

So the resolution of this defect is now be possible simply by setting the default values of the `editor.scrollbar.horizontalScrollbarSize` and `editor.scrollbar.verticalScrollbarSize` settings to 14