arduino/arduino-ide #437: Cursor randomly jumps to beginning of line

tomotvos, 2021-07-06:
**Describe the bug**
While editing, the cursor sometimes randomly jumps to the beginning of a new line

**To Reproduce**
Steps to reproduce the behavior:
1. Edit sketch
2. Usually when typing a new line, cursor will jump to beginning of next line
3. Seems to be related to auto-indent with tabs

**Expected behavior**
The cursor should not jump around and stay where I am typing

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: Big Sur
 - IDE: 2.0.0-beta.7
Date: 2021-05-17T10:14:31.139Z
CLI Version: 0.18.3 alpha [d710b642]

**Additional context**
Add any other context about the problem here.

ubidefeo, 2021-07-07:
@tomotvos 
this is a bug we have experienced but had not made it into the backlog yet, so thank you for reporting it.
It is caused by the Language Server pre-processing which we'll get to in the future to make it less prone to interfere with the user's cursor. After all getting the cursor stolen from you as you type is not something anyone can like 👍🏼

gdarthy, 2021-11-07:
It happens everytime you are on new line, then press TAB, cursor jumps back to home position of line, then, when you type anything, it jumps one line below. It's very annoying, I don't understand why this is not fixed yet since it makes working with IDE v2 frustrating. Considering going back to 1.8. 

Bug still on IDE 2.0.0-beta.12 on Windows 10

ubidefeo, 2021-11-07:
@gdarthy this bug is due to a faulty interaction between the editor and the Language Server.
We're rewriting the LS to address a lot of these bugs.
We understand the frustration, it bothers us as much as it bothers you :)

The team is hard at work to release a stable IDE 2.0

Thank you for your comment.
Ubi

MArimont3, 2022-02-14:
IDE 2.0 rc3 (Windows 10) : cursor jumps after hitting enter -> jumps to new line, removes spaces before previous line, breaks words, inserts spaces,  ... 
Very weird behavior.
I am using the IDE since end of dec. Seems to get worse and worse. Becomes nearly unusable...

ubidefeo, 2022-07-29:
@cmaglie @per1234 
can we close this one?

cmaglie, 2022-07-29:
> can we close this one?
No, it's still happening with the current nightly:

Version: 2.0.0-rc9-nightly-20220729
Date: 2022-07-29T03:40:20.747Z
CLI Version: 0.25.0 [4fd95834]

Willem43T, 2022-08-10:
I am experiencing this with the latest nightly.  I did not have this before, only with the latest few nightlys.

Version: 2.0.0-rc9.1-nightly-20220810
Date: 2022-08-10T09:41:17.361Z
CLI Version: 0.0.0-git []

per1234, 2022-08-20:
I am able to reproduce this on my Linux machine, but not on my Windows machine.

@Willem43T did some investigation and shared their valuable findings on the Arduino Forum:

https://forum.arduino.cc/t/cursor-jumping-problem/1023535

I'll document what I have learned about the issue here:

### To reproduce

1. Select **File > Preferences...** from the Arduino IDE menus.
1. Check the box next to "**☐ Auto save**".
   **ⓘ** I use **Auto save** for this demo because it produces the more disruptive behavior as reported in the issue, but the bug occurs even on manual save when Auto Save is disabled.
1. Click the <kbd>**OK**</kbd> button.
1. Open the "**Command Palette**" via the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) keyboard shortcut.
1. Select the "**Preferences: Open Settings (UI)**" command.
1. In the "**Search Settings**" field, type `editor.trimAutoWhitespace`
1. Check the checkbox next to "**☐ Remove trailing auto inserted whitespace.**"
   **ⓘ** This is the default, the instructions are only ensuring you don't have a custom setting.
1. In the "**Search Settings**" field, type `files.trimTrailingWhitespace`
1. Uncheck the checkbox next to "**`☑` When enabled, will trim trailing whitespace when saving a file.**"
   **ⓘ** This is the default, the instructions are only ensuring you don't have a custom setting.
1. Create the following sketch code:
   ```cpp
   void setup() {
     // indented line
   }
   void loop() {}
   ```
1. Place the cursor at the end of line 2.
1. Press <kbd>**Enter**</kbd>
   🙂 Whitespace is automatically inserted at the start of the new line to indent it to the same level as the previous one:
   [image]
1. Wait for the Auto Save operation to complete, as indicated by the disappearance of the **⬤** indicator from the right side of the sketch tab.

🐛 The whitespace is removed from line 2:

[image]

**ⓘ** I set `"editor.renderWhitespace": "all"` for these screenshots in order to make the whitespace visible as a "•".

### Expected behavior

In short: the VS Code behavior. This is the behavior the Arduino IDE has on my Windows machine.

I expect the Arduino IDE to work exactly like that, and to work that way consistently.

I'll provide the exact descriptions of the expected behavior of each of the related settings.

#### `editor.trimAutoWhitespace`

**Default:** `true`

Remove automatically inserted whitespace when it would end up as trailing whitespace.

This feature is completely unrelated to saving. Trimming is done on the fly while the user is editing, even when auto save is disabled.

1. Open the "**Command Palette**" via the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) keyboard shortcut.
1. Select the "**Preferences: Open Settings (UI)**" command.
1. In the "**Search Settings**" field, type `editor.trimAutoWhitespace`
1. Check the checkbox next to "**☐ Remove trailing auto inserted whitespace.**"
1. Create the following sketch code:
   ```cpp
   void setup() {
     // indented line
   }
   void loop() {}
   ```
1. Place the cursor at the end of line 2.
1. Press <kbd>**Enter**</kbd>
[... 65 lines cut ...]
I am not able to reproduce the bug on my Windows machine using **Theia Blueprint**.

This leads me to believe that the bug is in [**Eclipse Theia**](https://github.com/eclipse-theia/theia) or one of its dependencies, not a bug in the Arduino IDE code base.

---

I found this similar issue in the Theia repository: https://github.com/eclipse-theia/theia/issues/8718

However, that is related to the presence of [an `.editorconfig` file](https://editorconfig.org/). The bug occurs in sketches that do not contain such a file and I also don't have any in the parent folders. I also tried adding one to the sketch with `trim_trailing_whitespace = false`, and the bug still occurs. This leads me to believe that the bug is different from that described in https://github.com/eclipse-theia/theia/issues/8718.

---

The fact that the issue occurs in **Theia Blueprint**, and that it occurs even after I run the "**Arduino: Stop Language Server**" command or when I have no board selected seems to be evidence against the previous hypothesis that this was related to the language server.

---

I am not able to reproduce the bug using VS Code with the same settings on my Linux machine.

---

As discovered by @Willem43T, the bug no longer occurs if `editor.trimAutoWhitespace` is set to `false`. However, I think this feature is useful when working as intended and also provides parity with Arduino IDE 1.x, which has equivalent behavior.

---

The behavior of the bug does not match the behavior of `files.trimTrailingWhitespace`. Only the automatically inserted whitespace is removed. Any other trailing whitespace in the sketch is preserved.

KurtE, 2022-08-28:
As I mentioned in 1364 this is one of a few bugs that would keep me from using IDE2 for any serious stuff.
The other main one is the functionality of the Serial Monitor.  

Hope it gets fixed soon.  Will try the workaround.  But as mentioned I don't think that is the correct end solution, as I would still prefer the files to have trailing white space removed.

Willem43T, 2022-08-29:
The cursor jump bug made the editor mostly unusable for me. With the workaround I can use IDE 2 which offers huge advantages for me and the way I work.  I agree it is not ideal and definitely not the final solution.  However, since the complier does not mind a few possible trailing spaces, neither do I (for the moment).  I am sure it will eventually be solved.

The serial monitor is a different issue and not one that particularly effects me at present - I am sure it will also eventually be solved.

Considering the progress over the last few updates, the team is doing well - keep up the good work.

KurtE, 2022-08-29:
@Willem43T - I totally agree, the work around, helped a lot.  Keeps me from using another editor to make the changes.  And I agree that for the most part, things keep getting better

Suggestion to Arduino: 
How are users expected to even know there are additional preferences that can be changed?
That is:
```
Open the "Command Palette" via the Ctrl+Shift+P (Command+Shift+P for macOS users) keyboard shortcut.
Select the "Preferences: Open Settings (UI)" command.
In the "Search Settings" field, type editor.trimAutoWhitespace
Check the checkbox next to "☐ Remove trailing auto inserted whitespace."
```
Maybe a Menu item somewhere?  Tools->Advanced->Command Palette

Maybe something at the end of the preferences dialog, like the Arduino 1:
```
More preferences can be edited directly in the file:'
c:\Users\kurte\appData\local\arduino15\preferences.txt
...
```
Where, maybe instead button that takes you to Command Palette or to the UI preferences page...

Serial Monitor - I use a lot, to help debug things as well as to document stuff.   So, hopefully you are right, and they will fix the major issues with it soon.

Willem43T, 2022-08-29:
@KurtE - F1 will open the Advanced Preferences (same as Ctrl+Shift+P). In search enter "settings" and select "Open Settings (UI)" from the list.

Under "Editor" you now have hundreds of understandable as well as some really obscure settings. Quite a way down you will also find, under the "T"s (with Editor still highlighted on the left), the above "Trim Auto Whitespace".

Have fun.

kittaakos, 2023-03-21:
Hi, I have a proposed fix in https://github.com/arduino/arduino-ide/pull/1972. Could somebody help with the verification? [Here](https://github.com/arduino/arduino-ide/blob/main/docs/contributor-guide/beta-testing.md) is the documentation on how to do the beta testing. Thank you!

commit 2b17b2bf (2023-03-21): fix: let the resource finish all write operation

before checking if it's in sync or not.

Closes #437
PR #1972 (2023-03-28): fix: let the resource finish all write operation
### Motivation

As noted in https://github.com/eclipse-theia/theia/issues/12327, there could be a race condition when saving the editor model. This PR customizes the file resource resolver to create instances that wait for all running write operations to finish before checking whether the resource is in sync.

### Change description

### Other information

It's not trivial to verify the changes. Without this PR, the bug can be reproduced with [these](https://github.com/arduino/arduino-ide/issues/437#issuecomment-1221379270) steps.

https://user- [... cut]
commit 39ab8368 (2023-03-28): fix: let the resource finish all write operation

before checking if it's in sync or not.

Closes #437
closed 2023-03-28 by kittaakos