arduino/arduino-ide #443: Add `arduino_secrets.h` to Cloud sketches that contain secrets not defined via the web interface

per1234, 2021-07-08:
Arduino Web Editor has [a "Secret tab" feature](https://create.arduino.cc/projecthub/Arduino_Genuino/store-your-sensitive-data-safely-when-sharing-a-sketch-e7d0f0) that makes it easy to share sketches without accidentally exposing sensitive data (e.g., passwords or tokens). Secrets are automatically generated for any identifier in the sketch which starts with `SECRET_` and contains all uppercase characters.

The values of these secrets can be defined on Arduino Cloud via a form in the "Secret" tab in Arduino Web Editor.

Secrets are also used in Arduino IoT Cloud Thing sketches to store network credentials. These can be set via the Arduino IoT Cloud web interface.

When you download a sketch containing secrets from Arduino Web Editor, it includes a file named `arduino_secrets.h` that contains `#define` directives for the secrets.

When you have defined the value of secrets via the Arduino Cloud web interfaces and then pull the sketch via Arduino IDE's Cloud Sketchbook, the `arduino_secrets.h` file is added to the sketch. However, if you have not defined values for the secrets via the web interface before pulling, then the sketch will not contain this file.

**To Reproduce**
1. Steps to reproduce the behavior:
1. Open [Arduino Web Editor](https://create.arduino.cc/editor).
1. Click the **New Sketch** button.
1. Add the following code:
   ```cpp
   char myPassword[]=SECRET_PASSWORD;
   void setup() {}
   void loop() {}
   ```
   You should see a tab named "Secret" appear. Do not define the value of `SECRET_PASSWORD`.
1. Start the Arduino IDE.
1. Open the sketchbook explorer sidebar (folder icon)
2. Click the "Cloud Sketchbook" tab (globe icon)
3. Click the "Sign in" button and sign into your Arduino account if not already.
4. In the Arduino IDE, click the name of the sketch you created in Arduino Web Editor.
5. Click the "Pull" icon (cloud with downward arrow). Wait for the pull to finish.
6. Double click the sketch to open it. The sketch contains only the primary `.ino` file and `ReadMe.adoc`.

**Expected behavior**
When a sketch that uses secrets is pulled, it will always contain an `arduino_secrets.h` file with `#define` directives for the secret macros, with an `#include` directive at the top of the primary sketch file for `arduino_secrets.h` (https://github.com/arduino/arduino-ide/pull/438)

**Desktop (please complete the following information):**
 - OS: Windows
 - Version: Version: 2.0.0-beta.8-snapshot.491fa19
Date: 2021-07-07T15:45:28.645Z
CLI Version: 0.18.3 alpha [d710b642]

**Additional context**

I can envision two common use cases where secrets are not defined on Arduino Cloud before pulling the sketch:

- The Arduino IoT Cloud web interface is used for the initial setup of a Thing, but the user wants to do all further sketch development in the IDE.
- A copy of a sketch is created in Arduino Web editor via **⬤⬤⬤ > Save As** (secret values are not transferred to the saved copy of the sketch).

fstasi, 2022-01-12:
Addendum: we read secrets from the sketch json file and we create the `arduino_secrets.h` file manually when secrets exists.

I need to confirm this, but this might require a change on the arduino cloud api so that the "secrets" section in the json is always populated even if secrets are defined but have no assigned value

kittaakos, 2023-01-16:
If the value of a secret is not defined, the server is not returning with the `sketch.secrets` property.

I have two sketches in the Create editor:

`alma`:
```cpp
char myPassword[] = SECRET_PASSWORD;
void setup() {}
void loop() {}
```
No value is defined for the secret
[image]

`korte`:
```cpp
char myPassword[] = SECRET_PASSWORD;
void setup() {}
void loop() {}
```

For this sketch, there is a value defined for the secret:

[image]

1. List resources in the root:
```sh
curl \
  -H "Accept: application/json" \
  -H "Content-Type: application/json" \
  -H "authorization: Bearer $TOKEN" \
  "https://api2.arduino.cc/create/v2/files/d/\$HOME/sketches_v2" | jq .
```

```json
[
  {
    "children": 2,
    "href": "/create/v2/files/d/8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/alma",
    "modified_at": "2023-01-16T13:46:27.698Z",
    "name": "alma",
    "path": "8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/alma",
    "size": 1576,
    "sketch_info": {
      "id": "7258bf5a-e1ad-45ae-82bc-bfde42071cb7",
      "is_public": false
    },
    "type": "sketch"
  },
  {
    "children": 2,
    "href": "/create/v2/files/d/8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/korte",
    "modified_at": "2023-01-16T13:46:54.140329Z",
    "name": "korte",
    "path": "8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/korte",
    "size": 1576,
    "sketch_info": {
      "id": "8d1c721b-dd40-4ad6-8a95-a725032905f9",
      "is_public": false
    },
    "type": "sketch"
  }
]
```

2. Get the sketch details, a value for the secret is not defined:
```sh
curl \
  -H "Accept: application/json" \
  -H "Content-Type: application/json" \
  -H "authorization: Bearer $TOKEN" \
  "https://api2.arduino.cc/create/v2/sketches/byID/7258bf5a-e1ad-45ae-82bc-bfde42071cb7" | jq .
```

```json
{
  "board_fqbn": "arduino:avr:ethernet",
  "board_name": "Arduino Ethernet",
  "board_type": "serial",
  "created_at": "2022-10-25T14:02:12.360033Z",
  "href": "/create/v2/files/d/8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/alma",
  "id": "7258bf5a-e1ad-45ae-82bc-bfde42071cb7",
  "is_public": false,
  "libraries": [],
  "modified_at": "2023-01-16T13:46:27.698Z",
  "name": "alma",
  "path": "8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/alma",
  "size": 1723,
  "tutorials": null,
  "types": null,
  "user_id": "b036b0a3-ef0e-44b1-af37-3c756f20fdf0"
}
```

3. Get the sketch details when the value is defined for the secret:
```sh
curl \
  -H "Accept: application/json" \
  -H "Content-Type: application/json" \
  -H "authorization: Bearer $TOKEN" \
  "https://api2.arduino.cc/create/v2/sketches/byID/8d1c721b-dd40-4ad6-8a95-a725032905f9" | jq .
```

```json
{
  "board_fqbn": "",
  "board_name": "",
  "board_type": "",
  "created_at": "2023-01-16T10:57:24.474541Z",
  "href": "/create/v2/files/d/8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/korte",
  "id": "8d1c721b-dd40-4ad6-8a95-a725032905f9",
  "is_public": false,
  "libraries": [],
  "modified_at": "2023-01-16T13:46:54.140329Z",
  "name": "korte",
  "path": "8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/korte",
  "secrets": [
    {
      "name": "SECRET_PASSWORD",
      "value": "xxx"
    }
  ],
  "size": 1747,
  "tutorials": null,
  "types": null,
  "user_id": "b036b0a3-ef0e-44b1-af37-3c756f20fdf0"
}
```

The response object contains the `secrets` if the value is defined. Otherwise, it does not. Without the expected object, I do not know how IDE2 can fix this issue.

kittaakos, 2023-02-14:
From the browser's network tab, I can see that the request and response are correct when I leave the secret value empty and the auto-save does the update:

Request:
```
POST https://api2.arduino.cc/create/v2/sketches/8f7a44c7-98bf-4084-9ad7-5a100cf07ecb
```

Request payload:
```json
{
  "id": "8f7a44c7-98bf-4084-9ad7-5a100cf07ecb",
  "secrets": {
    "data": [
      {
        "name": "SECRET_PASSWORD",
        "value": ""
      }
    ]
  },
  "libraries": []
}
```

Response:
```json
{
    "board_fqbn": "",
    "board_name": "",
    "board_type": "",
    "created_at": "2023-02-14T12:35:08.436905Z",
    "href": "/create/v2/files/d/8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/copy_to_cloud",
    "id": "8f7a44c7-98bf-4084-9ad7-5a100cf07ecb",
    "is_public": false,
    "libraries": [],
    "modified_at": "2023-02-14T14:13:47.678774Z",
    "name": "copy_to_cloud",
    "path": "8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/copy_to_cloud",
    "secrets": [
        {
            "name": "SECRET_PASSWORD",
            "value": ""
        }
    ],
    "tutorials": null,
    "types": null,
    "user_id": "b036b0a3-ef0e-44b1-af37-3c756f20fdf0"
}
```

The problem must be with the GET part.