arduino/arduino-ide #448: The scroller and miniature view like in VS Code

CptHolzschnauz, 2021-07-11:
**Is your feature request related to a problem? Please describe.**
In long sketches you often need the scroller at the right and the thumbnail view  is used for orientation

**Describe the solution you'd like**
Like they do in Visual Studio Code. A miniature view (and a scroller)with increased width.

[image]

**Describe alternatives you've considered**
If this is a low hanging fruit that just needs to be enabled in theia i suggest to do so.

**Additional context**

Thanxs to all the contributors! This IDE is such an improvement!

CptHolzschnauz, 2021-07-12:
For other minibar lovers who found this thread: just change to
`"editor.minimap.enabled": true`
in bundle.js and index.js.
Too little for a pull request.

kittaakos, 2023-05-11:
> just change to
> `"editor.minimap.enabled": true`
> in bundle.js and index.js.

The minimap is disabled by default. One can use more convenient ways to enable the minimap:

1. _Command Palette_:
  - <kbd>Ctrl/⌘</kbd>+<kbd>Shift</kbd>+<kbd>P</kbd>,
  - type `View: Toggle Mininap`, and
  - press <kbd>Enter</kbd>.

[image]

2. Use [Advanced settings](https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#advanced-settings) UI:
 - <kbd>Ctrl/⌘</kbd>+<kbd>Shift</kbd>+<kbd>P</kbd>,
 - type `Preferences: Open Settings (UI)`,
 - type `minimap` in the search field, and
 - click the checkbox to have the minimap in IDE2.

[image]

3. Configure the advanced settings in the `settings.json`:
 - <kbd>Ctrl/⌘</kbd>+<kbd>Shift</kbd>+<kbd>P</kbd>,
 - type `Preferences: Open Settings (JSON)`, and
 - add the `"editor.minimap.enabled": true`.

[image]

```jsonc
{
  // other settings
  "editor.minimap.enabled": true`
  // maybe more settings
}
```

The minimap can be enabled from the primary UI without modifying any files inside the application. I am closing it as _won't do_.

If you disagree, please ping me, and we can discuss it.
If you maintainer, reopen the ticket.

Thanks!

closed 2023-05-11 by kittaakos