arduino/arduino-ide #451: ESP32 Doit Devkit v1 uploading Sketch(s) impossible

ZEROF, 2021-07-13:
### Hardware:
Board:						     ESP32 DOIT Devkit v1	    	
IDE name:					     Tested Arduino IDE (1.8.13 and 2.x beta)
Flash Frequency:				     Tested 40/80Mhz
Upload Speed:				     Tested 921600/115200
Computer OS:                                      Ubuntu 20.04, as well with Ubuntu 18.04/Kernel 5.4.0-77-generic

[image]

### Description:

Before (few months before) this was like riding a bike, connect board to PC, upload what you want, have fun but now it's impossible to upload anything to ESP32, even blink example. I tested with 5 USB cables and 3 ESP32 boards, just to be sure that I don't have some hardware related issue. I tried with holding "BOOT" button as well (never needed to do this), but it ends like this:

[image]

or

[image]

If i just try to upload without touching anything, just connecting board to pc (like it was possible before):

[image]

To be sure that board is well connected i try I don't know how many times, and from dmesg I can see (ports ttyUSB0, ttyUSB1, ttyUSB2 etc..)

[image]

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/Blink
*/

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}

```

### Debug Messages:
```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1252
load:0x40078000,len:12732
load:0x40080400,len:3100
entry 0x400805ec
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1252
load:0x40078000,len:12732
load:0x40080400,len:3100
entry 0x400805ec
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1252
load:0x40078000,len:12732
load:0x40080400,len:3100
entry 0x400805ec
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1252
load:0x40078000,len:12732
load:0x40080400,len:3100
entry 0x400805ec
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1252
load:0x40078000,len:12732
load:0x40080400,len:3100
entry 0x400805ec
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1252
load:0x40078000,len:12732
load:0x40080400,len:3100
entry 0x400805ec
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv�<� �W|�KW|�K�r�+��r�+��r�+��r����r�+��r����r����r������K�r�+O,
```

HilmarSt, 2021-07-24:
I had a similar issue with an ATmega328 while I was switching (not the first time) from the old IDE to test the new beta IDE.
In the boards manager, the board status was "installed" even if the board was not selectable in the tools menu.
I've "solved" the issue by installing an older board version in the boards manager and then re-installed the actual version again.

per1234, 2021-12-21:
Hi @ZEROF. Did you ever solve this issue?

Since you report getting it with two different versions of the Arduino IDE which are separated from each other by years and have very little shared code, it seems more likely to me that the cause of your problem is in the board hardware, USB cable, or ESP32 platform rather than the IDE.

ZEROF, 2021-12-21:
Hi @per1234 ,

That would be nice, but issue is not related to anything listed. I have 4-5 esp32 and esp8266 boards, 10 cables, 3-4 versions of IDE etc .... As I said I have Linux Laptop, Carbon X1 8th generation and I don't have USB 2.0. I made working some sketches on my old laptop x201. Changing speed, cables, IDE version did nothing. Last try was updating usb firmware, bios etc. on my "new" laptop. But I keep getting same errors or new ones.

I went to full debug, setting again permissions for user to use tty and putting non root user to dialout group etc...Nothing resolved this issue. It's like this boards are living in the past, and they are working much better for people having Windows or Mac, and that's is said.

per1234, 2021-12-21:
OK, I appreciate your quick response.

Since this problem is happening while the 3rd party [Espressif esptool](https://github.com/espressif/esptool) upload tool is in direct communication with the ESP32 board, I don't think the issue is related to the Arduino IDE so I'll go ahead and close this as off topic.

You can ask about this problem over on the Arduino Forum if you like. It is likely someone can provide advice over there:

https://forum.arduino.cc/

closed 2021-12-21 by per1234