arduino/arduino-ide #454: Can't see the hex files after exporting compiled binary

Carnoval15, 2021-07-15:
**Describe the bug**
The pro-IDE isn't generating the hex file!

**To Reproduce**
Steps to reproduce the behavior:
1. easily, make a project, use a simple code
2. click: "Export compiled binary"
3. Check the sketch folder
4. And now you can't see the hex file

**Expected behavior**
There must be 2 files with hex extension

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 21H1

Can anyone fix this? or help me if I do the wrong thing. Thx!

per1234, 2021-07-15:
Hi @Carnoval15. Thanks for taking the time to submit an issue.

The exported binaries are now located in the `build/arduino.avr.uno` subfolder of the sketch folder.

This was an intentional change, so the different location should not be seen as a bug, but we definitely need to document the changed behavior to allow a painless migration from the classic Arduino IDE to Arduino IDE 2.x, so the lack of such documentation (as far as I know) could certainly be considered a bug.

Carnoval15, 2021-07-15:
I see! I found the hex files in there.
thanks for helping!

closed 2021-07-19 by Carnoval15