arduino/arduino-ide #455: OSX Keyboard shortcut behaviour clashes with operating system shortcuts

DeeEmm, 2021-07-16:
**Describe the bug**
Keyboard shortcut behaviour clashes with operating system shortcuts

**To Reproduce**
Command+W shortcut in IDE closes project / application (arduino-close)

**Expected behavior**
Command+W shortcut in IDE should close active tab (as per OSX standard behavior)

Alternatively...

Command+Q shortcut in IDE should close application (as per OSX standard behavior)

**Desktop (please complete the following information):**
 - OS: OSX
 - Version: 11.4 Big Sur

**Additional context**
Shortcut key behaviour should follow operating system standard keys

kittaakos, 2021-07-16:
> Command+W shortcut in IDE closes project / application

Only if the [_active_ widget](https://github.com/eclipse-theia/theia/blob/44fd4051d0aa414aabb3913f7eb20246b78fd49b/packages/core/src/browser/shell/application-shell.ts#L872-L882) is not closeable. Otherwise, the _active_ widget will be closed and not the window. For example, if you look into the source code of your favorite lib by navigating into it, an editor tab will open. Press <kbd>Cmd</kbd>+<kbd>W</kbd>, and the editor will close. At least, the [implementation](https://github.com/arduino/arduino-ide/pull/1/commits/b34491269524eff0985fb624823c400f583c218a) was based on this idea.

> Command+W shortcut in IDE should close active tab (as per OSX standard behavior)

For the record, on macOS, if I do not have any active editors opened in VS Code and press <kbd>Cmd</kbd>+<kbd>W</kbd>, the window will be closed and not a tab. Long story short, I would not call it the standard macOS behavior.

DeeEmm, 2021-07-17:
> For the record, on macOS, if I do not have any active editors opened in VS Code and press Cmd+W, the window will be closed and not a tab. Long story short, I would not call it the standard macOS behavior.

Visual Studio is not a native OSX app. Being a Microsoft product It follows Windows conventions which are different from OSX conventions.

What you describe is partially true but native behaviour does not relate to _ACTIVE_ editors (tabs) it relates to  _VISIBLE_ tabs.

For example, when you close the last visible tab - for example on Safari, the app will close. But current behaviour on the IDE does not follow this as I still have visible tabs and pressing Cmd+W closes the app.

ubidefeo, 2022-07-29:
the shortcut closes the active sketch/window.
it's always been like this

closed 2022-07-29 by per1234