arduino/arduino-ide #461: Single letter Makro ('F') in ardware\avr\1.8.3\cores\arduino/WString.h

g2721, 2021-07-23:
**Describe the bug**
As far as I can tell, there is a Makro called 'F'  this file: _\Users\marcn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\cores\arduino/WString.h_

This prohibits the use of 'F' as a variable name on custom classes which are constructed with an parameter.

**To Reproduce**
Use a class with a constructor that looks like this:
```
class Example{
Example (int argument){}
};

//Then use the constructor:

Example F(42);
```

**Expected behavior**
Simple and common variable names shouldn't be used as makros by the underlying libaries. And in turn should be replaced by more unique names or not as Makros but consts.

**Screenshots**
[image]

**Error Code**

```
Arduino: 1.8.15 (Windows 10), Board: "Arduino Uno"

In file included from C:\Users\marcn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\cores\arduino/Arduino.h:28:0,

                 from sketch\sketch_jul23a.ino.cpp:1:

C:\Users\marcn\OneDrive\Dokumente\Arduino\sketch_jul23a\sketch_jul23a.ino: In function 'void setup()':

C:\Users\marcn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\cores\arduino/WString.h:38:74: error: initializer fails to determine size of '__c'

 #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))

                                                                          ^

C:\Users\marcn\OneDrive\Dokumente\Arduino\sketch_jul23a\sketch_jul23a.ino:6:11: note: in expansion of macro 'F'

   Example F(4);

           ^

C:\Users\marcn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\cores\arduino/WString.h:38:74: error: array must be initialized with a brace-enclosed initializer

 #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))

                                                                          ^

C:\Users\marcn\OneDrive\Dokumente\Arduino\sketch_jul23a\sketch_jul23a.ino:6:11: note: in expansion of macro 'F'

   Example F(4);

           ^

In file included from C:\Users\marcn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\cores\arduino/Arduino.h:232:0,

                 from sketch\sketch_jul23a.ino.cpp:1:

C:\Users\marcn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\cores\arduino/WString.h:38:28: warning: invalid conversion from 'const __FlashStringHelper*' to 'int' [-fpermissive]

 #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))

                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

C:\Users\marcn\OneDrive\Dokumente\Arduino\sketch_jul23a\sketch_jul23a.ino:6:11: note: in expansion of macro 'F'

   Example F(4);

           ^

C:\Users\marcn\OneDrive\Dokumente\Arduino\sketch_jul23a\sketch_jul23a.ino:2:1: note:   initializing argument 1 of 'Example::Example(int)'

 Example (int argument){}

 ^~~~~~~

In file included from C:\Users\marcn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\cores\arduino/Arduino.h:232:0,

                 from sketch\sketch_jul23a.ino.cpp:1:

C:\Users\marcn\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\cores\arduino/WString.h:38:95: error: 'Example::Example(int)' is private within this context

 #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))

                                                                                               ^

C:\Users\marcn\OneDrive\Dokumente\Arduino\sketch_jul23a\sketch_jul23a.ino:6:11: note: in expansion of macro 'F'

   Example F(4);

           ^

C:\Users\marcn\OneDrive\Dokumente\Arduino\sketch_jul23a\sketch_jul23a.ino:2:1: note: declared private here

 Example (int argument){}

 ^~~~~~~

exit status 1
Fehler beim Kompilieren für das Board Arduino Uno.
```

**Desktop (please complete the following information):**
 - OS: Win 10
 - Version: 2.0.0-beta 9 AND Version 1.8.15

**Additional context**
This issue is also present in the Arduino Ide 1.X.

per1234, 2021-07-23:
Hi @g2721. Thanks for your interest in this open source project.

This macro is a fundamental part of the Arduino core API. Even though you make a good point, that ship sailed over a decade ago. 
Feel free to `#undef` the macro as needed.

Anyway, this repository is for the code base of the Arduino IDE 2.x GUI, so discussion of the core library is off topic here.

closed 2021-07-23 by per1234

g2721, 2021-07-23:
Ok, fair enough. And thanks for the quick reply. I absolutely love the new IDE. 
-Marc