arduino/arduino-ide #463: Readme has a broken link.  Needs to be updated with correct file path

atlee19, 2021-07-24:
The Readme in `arduino-ide/arduino-ide-extension/` has a broken path link.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to  `arduino-ide/arduino-ide-extension/README.md` 
2. Scroll down to Monitor Service 
3. Click on the link `src/node/monitor-service-impl.ts `
4. It takes you to a 404 page not found

**Expected behavior**
I believe this file was moved into another folder named monitor, making the correct path for this file here `src/node/monitor/monitor-service-impl.ts`

**Proposed resolution**
Change this path `src/node/monitor-service-impl.ts`
To this `src/node/monitor/monitor-service-impl.ts`

yash-markad, 2021-08-02:
issue has been solved 
https://github.com/arduino/arduino-ide/pull/467#issue-701628116

fstasi, 2021-08-31:
Confirm this is solved. Thanks!

closed 2021-08-31 by fstasi