arduino/arduino-ide #465: Open Folder option produces error in Windows

Galbi3000, 2021-07-27:
In the Sketchbook view, if you select in the context menu for a Sketch "Open Folder" to open an Explorer window where the Sketch is stored you get the following error:

Windows cannot find '/c:/Documents/Arduino/MyProject'. Make sure you've typed the name correctly, then try again.

It should be obvious what is happening here for anyone observant enough. There is a leading forward slash which should not be there!

per1234, 2021-07-27:
Hi @Galbi3000. Thanks for taking the time to submit an issue.

This bug has already been fixed: https://github.com/arduino/arduino-ide/pull/450

You can give it a try by using the nightly build of Arduino IDE 2.x, which is available for download here:
https://www.arduino.cc/en/software#nightly-builds

closed 2021-07-27 by per1234

Galbi3000, 2021-07-27:
Ahh thanks for the info. I did not find #450 in my search for the issue because it did not match my search criteria, so apologies for the duplicate issue :)