arduino/arduino-ide #471: IDE v2.0.0 beta10 unuseable on macOS

sstaub, 2021-08-09:
**Describe the bug**
The last versions are complete unusable
- No Auto Complete
- No Outline
- No Errors
- Tab handling does not work, you can open a new Tab but can't close it
- .....

**To Reproduce**
Steps to reproduce the behavior:
Fresh install 

**Expected behavior**
All the functions I mentioned should work like in former releases

**Desktop (please complete the following information):**
 - macOS 11.5.1 BigSur

per1234, 2021-08-09:
Hi @sstaub. Thanks for taking the time to report the issues you experienced.

Please provide detailed instructions we can follow to reproduce the issue.

Please create a dedicated issue for each distinct class of issue. It looks to me like there are at least two independent problems you are reporting here. The first three might all be language server related (depending on what you mean by "errors", but surely the tab issue is unrelated. Mashing multiple issues together in one report makes it very difficult to track them, and thus more likely that one will be lost in the shuffle.

sstaub, 2021-08-09:
Just open the IDE and try to use the function I list, nothing happens.
- Maybe the language server is complete broken, so no Auto Complementation ....
I made also a fresh install, deleting all files in the system manually.
- e.g. Outline is complete empty
- When I create a new Tab, I can't delete or rename it
The IDE is big disaster, I don't understand why you don't make a plugin vor VSC.

per1234, 2021-08-09:
Please provide detailed instructions we can follow to reproduce the issue.

Please create a dedicated issue for each distinct class of issue. It looks to me like there are at least two independent problems you are reporting here. The first three might all be language server related (depending on what you mean by "errors"), but surely the tab issue is unrelated. Mashing multiple issues together in one report makes it very difficult to track them, and thus more likely that one will be lost in the shuffle.

sstaub, 2021-08-09:
After looking in my BootCamp WIN10 the IDE works as expected. When starting there was an indexing started for the language server. This doesn't happen in macOS. So I rename the Issue.

sstaub, 2021-08-09:
After deleting the hidden files .ArduinoIDE, the language is working as expected.

closed 2021-08-09 by sstaub