arduino/arduino-ide #474: Improvement of BUILDING.md (Contributions and development)

Wolfy7, 2021-08-12:
I would like to contribute to this project. Because i have this issue: https://github.com/arduino/arduino-ide/issues/385#issuecomment-895861208 i tried to build the project from source to analyse it. 

But im not a really familiar with TypeScript and the Theia IDE, so i faced some "issues" and open question.
As already requested in https://github.com/arduino/arduino-ide/issues/172 it would be helpfull if the Contributions and development process would be improved with the following (for example):

- how to debug with VS Code
for exapmle if i try to set a breakpoint, after starting the Debug "App (Electron)" the breakpoint changes to "Unbound breakpoint" want be reached: 
[image]
- Describtion of the diffrents Debug Configurations also for the diffrent tasks (launch.json and tasks.json)

Addditional idea(s):
- An area, in the forum or maybe use the GitHub-Wiki, for development/building/contributing were questions can be raised, known issues, best practices and so could be written down. To get in contact with more experienced contributers.

Happy Coding

per1234, 2021-09-12:
>An area, in the forum or maybe use the GitHub-Wiki, for development/building/contributing were questions can be raised, known issues, best practices and so could be written down. To get in contact with more experienced contributers.

There is already a dedicated section of the Arduino Forum for discussion of Arduino IDE 2.x:
https://forum.arduino.cc/c/software/arduino-ide-2-0-beta/93
Please feel free to use that.

Wolfy7, 2021-09-13:
> There is already a dedicated section of the Arduino Forum for discussion of Arduino IDE 2.x:
> https://forum.arduino.cc/c/software/arduino-ide-2-0-beta/93
> Please feel free to use that.

Thank you @per1234 that's what I was looking for, but didnt find.