arduino/arduino-ide #477: [20210814_Linux_64bit] Cannot create new sketch

nandanvasudevan, 2021-08-14:
**Describe the bug**
Not able to create a new sketch, the wntire GUI is non-responsive except for the open button.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `./arduino-ide` (downloaded from [arduino](https://downloads.arduino.cc/arduino-ide/nightly/arduino-ide_nightly-latest_Linux_64bit.zip))

**Expected behavior**
Create a sketch in the desired directory

**Screenshots**
NA

**Desktop (please complete the following information):**
 - OS: Linux - Ubuntu
 - Version: 20.04.2.LTS 64bit

**Additional context**
Add any other context about the problem here.

The IDE is non-responsive and the only option that works is open but cannot be accessed as it is behind the splash screen.

```bash
14/08/2021 11:22:22 nandanv ~/.../arduino-ide_nightly-20210814_Linux_64bit
└─ $ ▶  ./arduino-ide 
Starting backend process. PID: 15271
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root WARN please install @theia/electron@1.12.0-next.1a1d7afc as a runtime dependency
root INFO Configuration directory URI: 'file:///home/nandanv/.theia'
root INFO Theia app listening on http://localhost:34739.
root WARN The local plugin referenced by local-dir:/home/nandanv/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/home/nandanv/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:/home/nandanv/.arduinoProIDE/plugins does not exist.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.1-beta.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO Deploying backend plugin "cortex-debug@0.3.10" from "/home/nandanv/Downloads/arduino-ide_nightly-20210814_Linux_64bit/resources/app/plugins/cortex-debug/extension/dist/extension"
root INFO Deploying backend plugin "vscode-arduino-tools@0.0.1-beta.1" from "/home/nandanv/Downloads/arduino-ide_nightly-20210814_Linux_64bit/resources/app/plugins/vscode-arduino-tools/extension/dist/bundle"
root INFO Deploying backend plugin "cpp@1.52.1" from "/home/nandanv/Downloads/arduino-ide_nightly-20210814_Linux_64bit/resources/app/plugins/vscode-builtin-cpp/extension"
root INFO Deploying backend plugin "json@1.46.1" from "/home/nandanv/Downloads/arduino-ide_nightly-20210814_Linux_64bit/resources/app/plugins/vscode-builtin-json/extension"
root INFO Deploying backend plugin "json-language-features@1.46.1" from "/home/nandanv/Downloads/arduino-ide_nightly-20210814_Linux_64bit/resources/app/plugins/vscode-builtin-json-language-features/extension/client/dist/jsonMain"
root INFO Deploy plugins list took: 25.8 ms
root WARN Backend ConfigServiceImpl.onStart is slow, took: 194.5 ms
daemon INFO Starting daemon from /home/nandanv/Downloads/arduino-ide_nightly-20210814_Linux_64bit/resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
daemon INFO Using config file: /home/nandanv/.arduinoIDE/arduino-cli.yaml
daemon INFO arduino-cli version 0.18.3
daemon INFO Setting up Prometheus metrics on :9090/metrics
daemon INFO Starting daemon on TCP address 127.0.0.1:50051
daemon INFO Daemon is now listening on 127.0.0.1:50051...
daemon INFO Daemon is running.
root WARN Backend AuthenticationServiceImpl.onStart is slow, took: 94.5 ms
root INFO Git askpass helper is listening on http://127.0.0.1:39155.
root INFO Using Git [2.25.1] from the PATH. (/usr/bin/git)
root WARN Frontend CommonFrontendContribution.configure is slow, took: 161.8 ms
root WARN Failed to resolve module: @types/body-parser
root WARN Failed to resolve module: @types/cookie
root WARN Failed to resolve module: @types/express
root WARN Failed to resolve module: @types/fs-extra
root WARN Failed to resolve module: @types/lodash.debounce
root WARN Failed to resolve module: @types/lodash.throttle
root WARN Failed to resolve module: @types/react
root WARN Failed to resolve module: @types/react-dom
root WARN Failed to resolve module: @types/react-virtualized
root WARN Failed to resolve module: @types/route-parser
root WARN Failed to resolve module: @types/safer-buffer
root WARN Failed to resolve module: @types/ws
root WARN Failed to resolve module: @types/yargs
root WARN Failed to resolve module: @theia/monaco
root WARN Failed to resolve module: @theia/editor
root WARN Failed to resolve module: @types/rimraf
root WARN Failed to resolve module: @types/tar-fs
root WARN Failed to resolve module: @types/uuid
root WARN Failed to resolve module: @theia/navigator
root WARN Failed to resolve module: @types/p-queue
root WARN Failed to resolve module: @theia/preferences
root WARN Failed to resolve module: @theia/electron
root WARN Failed to resolve module: @theia/keymaps
root WARN Failed to resolve module: @theia/messages
root WARN Failed to resolve module: @types/diff
root WARN Failed to resolve module: @types/dompurify
root WARN Failed to resolve module: @types/mime
root WARN Failed to resolve module: @types/request
root WARN Failed to resolve module: @types/atob
root WARN Failed to resolve module: @types/auth0-js
root WARN Failed to resolve module: @types/btoa
root WARN Failed to resolve module: @types/dateformat
root WARN Failed to resolve module: @types/deepmerge
root WARN Failed to resolve module: @types/glob
root WARN Failed to resolve module: @types/google-protobuf
root WARN Failed to resolve module: @types/js-yaml
root WARN Failed to resolve module: @types/keytar
root WARN Failed to resolve module: @types/ncp
root WARN Failed to resolve module: @types/node-fetch
root WARN Failed to resolve module: @types/ps-tree
root WARN Failed to resolve module: @types/react-select
root WARN Failed to resolve module: @types/react-tabs
root WARN Failed to resolve module: @types/sinon
root WARN Failed to resolve module: @types/temp
root WARN Failed to resolve module: @types/which
root WARN Failed to resolve module: temp
daemon INFO Checking if CLI is Bundled into the IDE
daemon INFO Loading hardware from: /home/nandanv/.arduino15/packages
daemon INFO Adding libraries dir [dir: /home/nandanv/Arduino/libraries, location: user]
```

nandanvasudevan, 2021-08-14:
I have tested it on Windows 10, it is okay.

ConnahGriffin, 2021-10-05:
I'm currently using Ubuntu 20.04.3. It works just fine for me

nandanvasudevan, 2021-10-06:
_ add my user to the appropriate group_
I'm not quite sure what this means. Can you please explain that?

ConnahGriffin, 2021-10-07:
Oh, sorry for being vague. The dialout usergroup. So that I can properly access the usb with the IDE. Otherwise it would require the IDE to be run in `sudo`. There's also the input usergroup. This one if for systemd does practically the same thing.

per1234, 2022-05-29:
Hi @nandanvasudevan. Thanks for your report.

There have been some significant advancements in the Arduino IDE in the meantime. Please check to see whether the issue still occurs with the latest nightly build. The download links are here:

https://github.com/arduino/arduino-ide#nightly-builds

After that, let us know what the results were.

per1234, 2022-07-10:
Closing due to lack of response.

@nandanvasudevan if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

closed 2022-07-10 by per1234