arduino/arduino-ide #478: [2.0.0 B10] Better contrast for icons in Dark (Theia) theme

nandanvasudevan, 2021-08-14:
**Is your feature request related to a problem? Please describe.**
In Dark (Theia) theme, the top toolbar icons could use better contrast.

**Describe the solution you'd like**
Consider a lighter fill colour like the gray used for the other icons or maybe even white.

**Describe alternatives you've considered**
I am not sure, none come to mind.

**Additional context**
[image]
The icons are really hard to view when I have my blue light filter ON. Even otherwise they can surely use more contrast.

nandanvasudevan, 2021-08-14:
Just found developer tools and set it to white. I find this to be much better. Goes better with the theme too.
[image]

ubidefeo, 2022-07-29:
with our own themes being implemented and styled this issue seems irrelevant.
Please confirm @91volt

91volt, 2022-07-29:
Yes, I confirm that Arduino Dark theme provide an accessible contrast in toolbar actions. 
This issue can be considered solved for me
[image]
.

closed 2022-07-29 by per1234