arduino/arduino-ide #482: The file <file> has been changed on the file system

DaveTur, 2021-08-16:
In the Arduino IDE v2.0.0 beta 10, a window titled "The file <filename> has been changed on the file system"  popped up after each keystroke while editing a source file.  

This asked the question "Do you want to overwrite the changes made to <file> on the file system?".  Answering "Yes" caused my edits to be lost, choosing "No" allowed one further keystroke, before popping up again.  This also popped up when attempting to manually save the file.  The file was not being auto saved to the file system, despite "Auto Save" being enabled.  

After turning Auto-save off:  The pop-up no longer appeared after each keystroke. It did appear when attempting to manually save, by answering "Yes" the edits were saved to the file system.  Further manual saves (with or without making a change) updated the file's timestamp without the pop-up appearing, but any changes were not saved!

Turning auto-save back on:  The pop-up no longer appeared after each keystroke, the original auto-save behavior looked to have returned, a white circle being briefly shown, and the timestamp on the file system was updated, but again, the file's contents were not updated.

This behavior started after leaving the IDE open when my Windows 10 (64 bit) PC went to sleep.  I have subsequently rebooted the PC but the IDE's behavior continued.

My IDE had 31 files open with tabs in the 2nd screen row (many still there after looking at various library files).  The file currently being edited was shown twice on the filename tabs, but with different capitalisation.  The first letter of the filename was originally lower case, but this was renamed to upper case a week or so ago.  Edits made when the old filename (lower case) was selected are also seen when the new (upper case) file was selected, and vice versa.   The pull-down list of files (down arrow on right of file tabs) only showed the current upper case filename.  The old lower case file no longer existed anywhere on my PC, but a full search of my C: drive showed one "Internet Shortcut" to the lower case filename (in the project's folder on the D: drive).  The location of this shortcut is unknown.

After closing the tab of the old (lower cased) filename, then editing the current file (with upper cased filename), the behavior seems to have returned to normal.

per1234, 2022-05-29:
I am still able to reproduce this bug using the latest build (2.0.0-rc6-snapshot-522a5c6)

I'll provide instructions for reproducing it:

1. Select **File > New** from the Arduino IDE menus.
1. Click the <kbd>**▼**</kbd> button on the right side of the editor toolbar.
1. Select "**New Tab**" from the menu.
1. In the "**Name for new file**" field, type `Foo.ino`
1. Click the <kbd>**OK**</kbd> button. <br />
1. Select **File > Save As...** from the Arduino IDE menus.
1. Save the sketch to any convenient location.
1. Select **File > Close** from the Arduino IDE menus.
1. Use your file browser or terminal software to rename the file `Foo.ino` in the sketch folder to `foo.ino`.
1. Open the sketch in the Arduino IDE.
   🐛 The editor still contains a "**Foo.ino**" tab in addition to the expected "**foo.ino**" tab.
1. Select **File > Preferences** from the Arduino IDE menus.
1. Check the box next to "**☐ Auto save**".
1. Click the <kbd>**OK**</kbd> button.
1. Add some text to the "**foo.ino**" editor tab:
   ```cpp
   // hello world
   ```
1. Select the "**Foo.ino**" editor tab.
   😕 The editor does not contain the text that was added to the "**foo.ino**" editor tab.
1. Type a character in the "**Foo.ino**" editor tab.
   🐛 An unexpected "**The file 'Foo.ino' has been changed on the file system.**" dialog opens.
1. Click the <kbd>**No**</kbd> button.
1. Type a character in the "**Foo.ino**" editor tab.
1. Click the <kbd>**Yes**</kbd> button in the dialog.
1. Select the "**foo.ino**" editor tab.
   🐛 The previous `// hello world` contents of the file have been replaced by the characters that were added to the "**Foo.ino**" editor tab.
1. Select **File > Preferences** from the Arduino IDE menus.
1. Uncheck the box next to "**`☑` Auto save**"
1. Click the <kbd>**OK**</kbd> button.
1. Add some text to the "**foo.ino**" editor tab:
   ```cpp
   // foobar
   ```
1. Select **File > Save** from the Arduino IDE menus.
   🐛 An unexpected "**The file 'Foo.ino' has been changed on the file system.**" dialog opens.
1. Click the <kbd>**Yes**</kbd> button.
   🐛 An unexpected "**The file 'foo.ino' has been changed on the file system.**" dialog opens.
1. Click the <kbd>**Yes**</kbd> button.
1. Select the "**foo.ino**" editor tab.
   😕 The editor does not contain the text that was added to the "**Foo.ino**" editor tab.

Lorandil, 2022-12-31:
I'm experiencing a similar problem:
If I press "CTRL+S" the file is saved and immediately a dialog appears stating that the file has changed on disk.
If I press "yes" the file in the editor and on the disk(!) are reverted to the original state.
Toggling the auto save option doesn't seem to change the behaviour.

I'm using the current v2.0.3 - I can retry with the latest nightly build if you think the problem has been fixed.

Lorandil, 2023-01-01:
Hi @per1234,
thanks for the workaround.
In the meantime I tried to rename my "Foo.h" to "foo.h" and changed the mixed includes of "Foo.h"/"foo.h" to be all "foo.h". 
That happens when you are using Windows ;)
Let's see if that fixes the problem for me.

Lorandil, 2023-01-02:
To make it short - streamlining the file names didn't help.

I used the workaround @per1234 mentioned and that fixed the issue for now.

HattonLe, 2024-06-25:
This problem is marked as type imperfection but it is more serious than that. I have noticed several times that the IDE fails to build a project if you for example in my case decide to refactor (and rename) a class. For example I had been building with a class X10ex.cpp, X10ex.h and then decided to refactor it to X10.cpp, X10.h using external tools \ file manager. Relaunching Arduino IDE it picks up the refactored files in the edit tabs... this is good... but when I do a build the compiler  comes up with errors in the original X10ex.cpp file...  this is bad! Relaunching IDE \ recompiling doesn't fix the fault.
When i was using Windows I used to find the temp folder where the IDE was building from and delete the out of date files myself, but in Linux I haven't found where these are as yet.
I'm running Ubuntu 22.04 LTS, and Arduino in a snap, Version 2.3.2, Date 2024-02-20T09:54:08.669Z (4 months ago) CLI version 0.35.3.
In the attached screenshot, the file X10ex.cpp doesn't exist in my source folder... its referring to the old file name prior to refactoring.

[image]

HattonLe, 2024-06-25:
I found the Arduino IDE temporary build folder... see attached screenshot for its contents.
If I manually delete the offending files via a terminal command "rm X10ex.*"  I can then successfully build the sketch using the Arduino IDE.
[image]