arduino/arduino-ide #490: Change the default port: `/dev/ttyS0->/dev/ttyUSB0`

nagayev, 2021-08-30:
Hello. I have tried to upload my sketch to UNO board and I coudn't.
So I had to run dmesg and then I see that arduino recognised as ttyUSB0 device, not as default ttyS0.
So I suppose to change the default port to USB. It's actual espessialy for boards whose connected using USB like my UNO.

ckuethe, 2022-01-25:
So much this. Either `ttyUSB0` or `ttyACM0`are likelier to be arduino devices than `ttyS0`

per1234, 2022-02-03:
Possibly related: https://github.com/arduino/serial-discovery/issues/37

per1234, 2022-11-07:
Arduino IDE 2.x does not have default ports. By default, no port is selected. If you select a board and port, then that selection is associated with the sketch. This might give the impression that a default port is coded into the IDE, but it is actually only the port you selected, which can be changed by selecting a different port.

The possibly related issue I mentioned above https://github.com/arduino/serial-discovery/issues/37 has been resolved.

closed 2022-11-07 by per1234