arduino/arduino-ide #499: arduino ide ver 1.x & 2.x connt open

tqangxl, 2021-09-02:
install or zip all version  is same

per1234, 2021-09-03:
Hi @tqangxl. We're going to need a lot more information from you if we are to investigate the issue.

This is the issue tracker for Arduino IDE 2.x, so we will focus on that version of the IDE (though the information that 1.x also doesn't start is useful).

First of all, you must download the latest nightly build of the IDE so we can be sure the problem hasn't already been resolved. You can get the nightly here:
https://github.com/arduino/arduino-ide#nightly-builds

Now please let us know if the problem still occurs when using the nightly build.

If yes, we need to know which operating system you are using.

per1234, 2021-09-08:
Closing due to lack of response.

@tqangxl if you'll provide the requested information I'll be happy to re-open this issue.

closed 2021-09-08 by per1234