arduino/arduino-ide #501: Command `config add` for additional URLs does not check presence of URL in `board_manager` tree

ubidefeo, 2021-09-08:
**Describe the bug**
When manually adding an additional url via `arduino-cli config add ...` the command does not check for existing matching entries in the list, hence generating multiple entries for the same URL

**To Reproduce**
Steps to reproduce the behavior:
1. `arduino-cli config add board_manager.additional_urls https://my.url/package_mycustom_index.json`
2. `arduino-cli config dump`
3. verify that our URL is at the end of the `board_manager` group of `additional_urls`
4. `arduino-cli config add board_manager.additional_urls https://my.url/package_mycustom_index.json`
5. `arduino-cli config dump`
6. verify that the URL is twice at the end of the `board_manager` group of `additional_urls`
7. `arduino-cli config remove board_manager.additional_urls https://my.url/package_mycustom_index.json`
8. `remove` will remove any istance of said URL

**Expected behavior**
The command should check for the presence of said URL and not add it if already there

**Screenshots**
n/a

**Desktop (please complete the following information):**
 - OS: Mac OS
 - Version: 11.5.2

**Additional context**
n/a

per1234, 2021-09-08:
Hi @ubidefeo. Shouldn't this issue be in the `arduino/arduino-cli` repository?

silvanocerza, 2021-09-08:
It certainly should. Moving it.