arduino/arduino-ide #503: Board type lost when changing usb port

moth4017, 2021-09-08:
Windows 10 pro, Arduino V2 Beta 11
If you have a sketch open board type selected with com port , you then unplug  Board from usb port , replug into a diffren port.
you ten select new port detected but  the board type is also lost , you then have to reselect board type.
\
[image]

[image]

ubidefeo, 2022-07-29:
@per1234 
I believe this is not an issue anymore.
I have tested on Mac OS and it keeps on finding the board no matter the physical port.
I'm also testing with an ESP32 which has no assigned VID:PID

per1234, 2022-07-30:
@ubidefeo I tried it out and found that the board/port association is per-sketch. This means it must be repeated every time you create a new sketch or open a sketch where the association process hasn't been performed.

I'll provide instructions to reproduce it:

### To reproduce

#### Equipment

- Any device that creates a serial port which is not identifiable as an Arduino board.
  For example:
  - classic Nano
  - FTDI FT232R module or cable
  - ESP8266 board
  - ESP32 board

#### Steps

1. Connect the board to your computer.
1. Select the port of the board from the **Board Selector** menu on the toolbar.
1. If the board/port association has not already been made, the "**Select Board**" dialog will open, as expected. Select the board from the "**BOARDS**" menu.
1. Click the <kbd>**OK**</kbd> button.
1. Open the **Board Selector** menu on the toolbar.
   🙂 The board and port are now selected in the **Board Selector** menu, as expected.
1. Unplug the board from your computer.
1. Connect the board to your computer.
1. Open the **Board Selector** menu on the toolbar.
   🙂 The board and port are selected in the **Board Selector** menu, as expected.
   [image]
1. Unplug the board from your computer.
   ❗ This step is mandatory because the board/port association of a connected board would transfer to the new sketch created in the next step.
1. Select **File > New** from the Arduino IDE menus.
1. Connect the board to your computer.
1. Open the **Board Selector** menu on the toolbar.
   😕 The port is listed as "**Unknown**" in the **Board Selector** menu. It will be necessary to repeat the board selection via the "**Select Board**" dialog after selecting the port from the menu.

[image]

### Arduino IDE version

2.0.0-rc9-snapshot-55927ac

### Operating system

Windows 10

jpedrick, 2022-10-27:
With the LILYGO T5 EPD 4.7" V2.3 I see similar behavior. Every time I reset the board, the serial connection is dropped. In the UI board selection dropdown a random board is associated with the serial port. If I click the serial port in the drop down it actually changes to that board. So, I have to go into "Select other board and port...", which then *usually* has my previous board selected and then select the serial port.

I think it would be simpler to have the serial port and board selection separated. I think users are unlikely to switch architectures/boards in rapid fashion. So, a simple dropdown for "board" and a separate dropdown for "port" would be perfect.

Another solution would be to set the board architecture in the .ino project file. In my case, I'm working on a project that is highly coupled with the board type. I'd like to be able to specify that.