arduino/arduino-ide #504: Variables in the 'Watch' window of the Debugger do not update as the code is single stepped.

DaveTur, 2021-09-13:
To see new values of the watched variables, that variable has to be added again to the Watch window.

The target Arduino Zero was connected to my PC using a Segger Mini EDU.
I set a break-point in the function of interest, which performs a shallow copy of an object.  After hitting the break-point, the source and destination objects were displayed in the Watch window, the source showing expected values and the destination showing random uninitialised values.  While single-stepping the code, the destination values did not change.  Setting a further break-point on the function exit, the code was run until this second break-point was hit.  The destination values were still unchanged in the watch window.  After again adding the destination object to the Watch window, the expected correct values were seen, while the values in the destination object that was added to the Watch window before the copying still showed the old random values. 

I expect to see the destination values change as the code is either single-stepped or when it stops after being run to the exit break-point.

My PC is running Windows 10 Pro 64 bit.
The Arduino IDE is 2.0.0-beta. 11

DaveTur, 2021-09-18:
During a later debug session, I see that the value of simple Watched variables ARE seen to update when changed by code.  Previously I had the name of a class instance in the watch window and it was the values of its members that were not being updated in the watch window.

per1234, 2023-01-04:
Closing due to lack of response.

@DaveTur if you ever find the time to make another test with the latest nightly build and find the problem still occurs, please comment here and I'll be happy to re-open this issue and try to make an investigation.

closed 2023-01-04 by per1234