arduino/arduino-ide #509: IDE Hanging on start on Win 10

BanksySan, 2021-09-19:
**Describe the bug**

I have installed the latest 2.x release `2.0.0-beta.11`.  When opening the application the splash screen initially displays, then the form behind it.
The splash screen does not disappear (until I tried to record the screen, then it disappeared).
The application is responsive to open the menu and clicking _File -> Open_ but not _File -> New_.
It usually has the spinning blue busy indicator displayed.

 
**To Reproduce**
Steps to reproduce the behavior:
1.  Download [executable](https://downloads.arduino.cc/arduino-ide/arduino-ide_2.0.0-beta.11_Windows_64bit.exe?_gl=1*8wgwpy*_ga*Njc1NzM3ODU4LjE2MzEyOTI1ODk.*_ga_NEXN8H46L5*MTYzMjA3Njc4OC4xMy4xLjE2MzIwNzc0MDYuMA..)
2. Install IDE
3. Run IDE

**Expected behavior**
IDE opens and I can edit/create sketches after the splash screen has gone.

**Screenshots**
See:  https://drive.google.com/file/d/1GRZ4sVXUNy8cOhavHuM8SfIsL4wLt9sE/view

**Desktop (please complete the following information):**
 - OS: Windows 10 Home, 
 - Version: 21H1 19043.1237

**Additional context**
System info:
**Processor**   Intel(R) Core(TM) i9-9880H CPU @ 2.30GHz   2.30 GHz
**Installed RAM**   32.0 GB (31.9 GB usable)
**System type** 64-bit operating system, x64-based processor
**Pen and touch**   Pen and touch support with 10 touch points

per1234, 2021-09-19:
Hi @BanksySan. Thanks for your report. I'm going to ask you to provide some additional information that might allow us to determine the cause of the problem.

Please do this:

1. Start File Explorer (AKA Windows Explorer).
1. Open the Arduino IDE's installation folder (usually `C:\Program Files\Arduino IDE` or `C:\Users\<user name>\AppData\Local\Programs\Arduino IDE`).
1. In the folder listing of the Arduino IDE installation folder, hold the **Shift** key while clicking the right hand button on the mouse.
1. Click "Open PowerShell window here". PowerShell will now open with the current directory set to the Arduino IDE installation folder.
1. Type `& ".\Arduino IDE"`.
1. Press **<kbd>Enter</kbd>**. Debug output should now be printed to the PowerShell window as the Arduino IDE starts (or fails to start).
1. Wait for the Arduino IDE to finish starting, or failing to start.
1. Switch back to the PowerShell window if the Arduino IDE window took the focus.
1. Right click on the title bar of the PowerShell window.
1. Click **Edit > Select All**. This will select all the text in the PowerShell window.
1. Press **<kbd>Ctrl</kbd>+<kbd>C</kbd>**. This will copy the contents of the PowerShell window to the clipboard.
1. In this GitHub issue thread, click on the reply field.
1. Press **<kbd>Ctrl</kbd>+<kbd>V</kbd>**. This will paste the output between the code tags.
1. Click the "Comment" button.

Hopefully that will give us some information we can use to help you with the problem.

BanksySan, 2021-09-20:
@per1234 I've just reinstalled it again and it loaded without problems.  If I can get it to happen again then I'll let you know.

The _fix_, if you can call it that was:

Day 1:

* Install via the `exe`.
* Run it (fails).
* Uninstall it.
* Install it via the MSI.
* Run it (fails).
* Uninstall it.

Day 2:

* Reinstall from the MSI with an intention of getting diagnostics.
* Run it (success).

per1234, 2021-09-20:
I'm glad to hear it's working now!

Since we won't be able to investigate the issue without the requested information, I'll go ahead and close this. You are welcome to comment here if you have more information and I'll re-open if necessary.

closed 2021-09-20 by per1234