arduino/arduino-ide #512: arduino-ide_2.0.0-beta.11_Windows_64bit.exe malware found during installation

PickyBiker, 2021-09-21:
When installing the title file, Webroot Secure Anywhere flags the following file as malware W32.Adware.Gen.

C:\Users\Mike\AppData\Local\Programs\Arduino IDE\resources\app\node_modules\drivelist\build\Release\drivelist.node
[image]

The install is on a PC with windows 10.

rsora, 2021-09-22:
Hi @PickyBiker thank you for your report!

I ran a quick scan on virus total using the Github link we provide as a source and the file is reported as clean
https://www.virustotal.com/gui/url/4dfe88ec1d9f89c4db275e276432b2a94304773a39f6dc9d87de2a433be6ac8e

Can you please share the exact link you used to download the IDE?
Was it the Github release page https://github.com/arduino/arduino-ide/releases
or the Arduino Swoftware page https://www.arduino.cc/en/software#experimental-software ? 

Thank you and happy coding!

PickyBiker, 2021-09-22:
Here is the link I used.

Hope it helps.
https://downloads.arduino.cc/arduino-ide/arduino-ide_2.0.0-beta.11_Windows_64bit.exe

wsmagee, 2021-10-07:
I had the same issue  I used the link from the https://www.arduino.cc/en/software page and selected the Win10 and newer 64 bit. 
https://downloads.arduino.cc/arduino-ide/arduino-ide_2.0.0-beta.11_Windows_64bit.exe

I am also using WebRoot and it does not like the drivelist.node file.  I had previously tried with V9 Beta and it did the same thing.
Thanks

jeffrbeyer, 2021-10-08:
I just sent both links through Virus Total and if came up clean. I also tried the install in a sandboxed vm and can see it does not drop that drivelist.node file anywhere , at anytime during installation. It might be picking it up from your system instead?