arduino/arduino-ide #518: Arduino IDE 2.0 beta silently fails if library example has .pde extension

lmtanco, 2021-09-28:
Hi there,

If the example for a library has .pde extension, Arduino IDE 2.0 beta will silently fail to open it.

To reproduce the issue:

- Install Arduino IDE 2.0 beta
- Search and Install TimerOne library from Arduino IDE 2.0 beta Library Manager
- Open Examples>Examples from Custom Libraries>TimerOne>Interrupt

Manually renaming the example file Interrupt.pde to Interrupt.ino solves the issue.

Tried this on MacOS. Version: 2.0.0-beta.11 Date: 2021-08-25T08:44:35.277Z (4 weeks ago) CLI Version: 0.18.3 alpha [d710b642]

On the [Arduino forum](https://forum.arduino.cc/t/arduino-ide-2-0-beta-silently-fails-if-library-example-has-pde-extension/909565), user [in0](https://forum.arduino.cc/u/in0/summary) has reproduced the error on Windows and provides additional info: 

> I can reproduce it. You can see the problem when run from the command line:

`root ERROR Request cloneExample failed with error: ENOENT: no such file or directory, rename 'c:\Users\<user name>\AppData\Local\Temp\.arduinoIDE-unsaved2021827-16716-11heo0w.vdua\Interrupt\Interrupt.pde' -> 'c:\Users\<user name>\AppData\Local\Temp\.arduinoIDE-unsaved2021827-16716-11heo0w.vdua\Interrupt\Interrupt.ino'`

> It's trying to rename the sketch in the temporary folder rather than in its true path under the library installation.

> I don't have this problem with a standalone sketch using the .pde file extension.

BTW I'm loving the new IDE. Many thanks for the good work! :)

Luis.

ubidefeo, 2021-10-05:
hi @lmtanco 
Thank you for the kind words :)

This issue is in our backlog, but since more and more 3rd party developers are updating their libraries and examples to comply with specifications, we gave it very low priority.
The `.pde` extension for Arduino has been long deprecated, and we feel that if we leave it as is nobody will care to bring their Sketch code into 2021 and beyond :)

I have created a PR to Paul with the extensions updated.
https://github.com/PaulStoffregen/TimerOne/pull/54

Thank you

mrkylegp, 2022-11-30:
I have this issue in 2.0.2. Can't open a .pde example for the PCM library. Very frustrating.

KurtE, 2022-12-02:
Just checked again, with yesterday's daily build (after 2.0.2) and still has issue.  

This time with the Radiohead library.

mrkylegp, 2022-12-02:
@KurtE Yep, it's very frustrating. Not exactly an upgrade IMO since several of the libraries I use have .pde files that wont open and the developers aren't going to be updating these libraries again. I'm back to using the older IDE for now.

nbourre, 2023-03-21:
My suggestion would be to detect if the library's examples are using the obsolete format, if so ask the user if he would like to upgrade them.

wh201906, 2023-05-02:
I think this has been fixed in commit 0ab28266dfe1859eb4e1c88502452f419b667099.
In the code before this, the `copy()` will rename the `.pde` into `.ino`, but it will try to access the new file with the old `.pde` extension, so it fails.
I have tested this bug on v2.1.0 and it was fixed.

lmtanco, 2023-09-29:
Apologies for not closing this before. 

This works perfectly now (v2.2.1). Even if the example has .pde extension, it will open it, then it will ask the user if they want to change its extension. 

Tested on macOS Apple Silicon (screenshot) and Windows 11.  

[image]

closed 2024-02-20 by per1234

drf5n, 2024-02-20:
This is broken in IDE 2.3.2.

For example, try opening the:

File/Examples/AccelStepper/Bounce example that refers to this file on my Mac:

    /Users/drf/Documents/Arduino/libraries/AccelStepper/examples/Bounce/Bounce.pde

Using the workaround for a differential test, rename your Random.pde to Random.ino within its library directory and the Arduino IDE will open it as usual.

For example:

    /Users/drf/Documents/Arduino/libraries/AccelStepper/examples/Random/Random.ino

...functions as expected.

Tested on MacOS Sonoma 14.2.1 on Apple Silicon M2 

[image]

No screen shot of the silent failure.

drf5n, 2024-02-20:
See https://github.com/arduino/arduino-ide/issues/2377 for 2.3.2 issue