arduino/arduino-ide #539: Problem Views

sstaub, 2021-10-09:
Handling problem could much better then now:
In the moment you have no overview about errors and warning like in VSCode with PlatformIO like
[image]
In the ArduinoIDE there are both items missing marked in the screenshot before.
In the moment error handling looks like this:
[image]
Detailed error message could only seen with right click.
Also with Clang it is possible to fix a problem very simple, in the screenshot you can see that there is fix available.
On the IDE like CodeRunner on Mac you have a button to do simple a fix:
[image]

sstaub, 2021-11-10:
In Commands "Toggle Problem View" open the window with detailed view like VSC.
Like other views the IDE doesn't remember after closing and restart.

per1234, 2021-11-10:
>Like other views the IDE doesn't remember after closing and restart.

We are already tracking that in your other issue: https://github.com/arduino/arduino-ide/issues/538