arduino/arduino-ide #540: Cannot manually access port number

gfmoore, 2021-10-09:
**Describe the bug**
I am using a usb hub for plugging in a nano (I think this is a clone!!!). The new ide doesn't recognise the board and I cannot set the port number (COM8) manually. The older IDE has no problem.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**

I should be able to see the Port number and select any port I want when I go to Select other boards and port

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Version: [e.g. 2.0.0]
Win 10 insiders

**Additional context**
Add any other context about the problem here.

per1234, 2021-10-09:
Hi @gfmoore. Thanks for your report!

Does the problem still occur if you close all Arduino IDE 2.x windows and then start the Arduino IDE 2.x again? I have had a problem like this as well, but for me it was intermittent, with discovery working most of the time, but not after seemingly random occasional starts of the IDE. The problem was not persistent, in that after a restart of the IDE discovery works (though I suspect it's really that I get another roll of the dice on each start up, and so it's possible yet statistically unlikely that the problem will occur on two subsequent starts).

per1234, 2021-10-11:
I'm glad to hear it discovered it!

>I've just reconnected

Even though it's good that we are able to work around the issue, I still hope that a real solution can be found. With that in mind, I am still interested in your report.

Would you mind clarifying what you mean by "reconnected"? Do you mean that you closed the Arduino IDE and then started it again (as has been my workaround), or do you mean that you unplugged your Nano from the computer/hub and then plugged it back in?

>I still would prefer an option to be able to physically select a port without discovery.

After years of using discovery (starting with the old Arduino IDE) without receiving much negative feedback on it, it seems to me that this approach is sufficient on its own so long as it is working reliably. I think it will be possible to get the new discovery system working just as well as the one from previous versions of the old IDE.

per1234, 2021-10-12:
Thanks for the clarification!
>what happens if I use multiple COM ports for other things

I think using the **Tools > Port** menu, just as you would with Arduino IDE 1.x, will be most intuitive in this case.