arduino/arduino-ide #549: Add a save output to file option to serial monitor

orjanv, 2021-10-13:
It would be very helpful to have the option to save the file content in serial monitor to a text file.

per1234, 2022-04-13:
There is also a request for this capability in Arduino IDE 1.x here: https://github.com/arduino/Arduino/issues/9479

mklemarczyk, 2022-06-24:
Would be great for long monitoring. To inspect the results the output is terrible, as you can not select and copy easily. 
More over I am worried about buffer that will get depleted when loggings many messages in 3h window.

Please make a way to store serial monitor into file. It does not need to be perfect. One write to file each 5 min is enough or write when we press the button.

DannyV1, 2023-01-17:
I Also would love to have this feature, I would use it daily with IDE V:2.03

jeremymatt, 2024-03-06:
This seems like pretty basic functionality for troubleshooting.  I'm still pretty new to arduino: I'm currently working on my third arduino-based project.  The lack of any way reasonable way to save more than about a page of serial monitor output (since [#1081](https://github.com/arduino/arduino-ide/issues/1081) has not been fixed) has caused me problems on two of those three projects.  

Is there any timeframe for when this will get implemented or #1081 will get fixed?  

Thanks for your work on this project.