arduino/arduino-ide #567: [DRAFT] Network available devices (ESP*, MKR) are only listed with IP. Hostname should be visible

ubidefeo, 2021-10-21:
**Describe the bug**
When deploying more than a single Local OTA enabled board, the `mdns-discovery` finds as many as available, and exposes the correct fields, but the `label` field is not transferred to / used by the IDE, hence listing the board as just an IP.
Once the board type is manually selected, it become associated.
(this connection seems to also be hard to break, needs investigation)

sample output of `mdns-discovery` for an ESP32 and a MKR1000
```
{
  "eventType": "remove",
  "port": {
    "address": "10.0.1.25",
    "label": "Arduino\\ MKR1000\\ v2 at 10.0.1.25",
    "protocol": "network",
    "protocolLabel": "Network Port",
    "properties": {
      ".": "mkr1000",
      "auth_upload": "yes",
      "board": "mkr1000",
      "hostname": "Arduino\\ MKR1000\\ v2.local.",
      "port": "65280",
      "ssh_upload": "no",
      "tcp_check": "no",
      "ttl": "4500"
    }
  }
}
{
  "eventType": "add",
  "port": {
    "address": "10.0.1.26",
    "label": "FeatherESP32 at 10.0.1.26",
    "protocol": "network",
    "protocolLabel": "Network Port",
    "properties": {
      "hostname": "FeatherESP32.local.",
      "port": "3232",
      "ttl": "120"
    }
  }
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Version: [e.g. 2.0.0]

**Additional context**
Reported in ESP8266 as https://github.com/esp8266/Arduino/issues/8344
Referenced in https://github.com/arduino/arduino-ide/issues/123

silvanocerza, 2022-02-01:
Closing this issue as we won't do it.

Recent versions of the `mdns-discovery` was returning ports that must not have been returned including those not supporting Arduino OTA. i.e. Printers, laptops and similar.

After arduino/mdns-discovery#23 is merged unsupported ports won't be shown anymore in the IDE.

me-no-dev, 2022-04-02:
> Closing this issue as we won't do it.
>
> Recent versions of the `mdns-discovery` was returning ports that must not have been returned including those not supporting Arduino OTA. i.e. Printers, laptops [...]

I am not sure why you will not do it, given that you need to only change which property of the `Port` structure is being shown in the list, but OK. Also, those are not unsupported boards. At least they were supported up until now. Are you dropping OTA support for all third-party boards?

per1234, 2022-04-02:
Hi @me-no-dev. I didn't see any mention of unsupported boards here.

"Unsupported ports" is in reference to random consumer devices (e.g., network connected printers) that were being discovered incidentally and cluttering the "ports" listing. The reason that is relevant is that it is more difficult for a user to find the correct port when there are additional irrelevant items in the listing.

There is no intention to filter the ports of 3rd party Arduino boards, and in fact Arduino has done a lot of work lately creating a general purpose [pluggable discovery system](https://arduino.github.io/arduino-cli/latest/pluggable-discovery-specification/) which will allow boards platform developers (both Arduino's and 3rd party) to make their hardware discoverable by the Arduino development software via any communication channel.

me-no-dev, 2022-04-02:
Hi @per1234 . I am commenting more or less on the issue's title and then the declination comment sounded like it will drop the above mentioned boards (which are good Arduino boards). I've been digging into the IDE and CLI the last few days in order to prepare ESP32-Arduino for the new features. The new discovery is what I am spending most time on. An issue I found can be seen [here](https://github.com/arduino/arduino-ide/issues/740#issuecomment-1086595394).

ubidefeo, 2022-04-04:
@me-no-dev @per1234 

I am convinced we should display a combination of label and IP, but the issue is that the ArduinoOTA library will default to `arduino` for board name, and that can be confusing.

We should resort to displaying the value of `label` coming out of the Discovery.

```json
{
  "eventType": "add",
  "port": {
    "address": "192.168.1.10",
    "label": "MyMKR_wifi_1010 at 192.168.1.10",
    "protocol": "network",
    "protocolLabel": "Network Port",
    "properties": {
      ".": "arduino",
      "auth_upload": "yes",
      "board": "arduino",
      "hostname": "Peppe.local.",
      "port": "65280",
      "ssh_upload": "no",
      "tcp_check": "no"
    }
  }
}
```

Conforming to the [Discovery Protocol specifications](https://arduino.github.io/arduino-cli/0.21/pluggable-discovery-specification/#list-command).
This will allow us to define the device name using the following syntax in ArduinoOTA

`ArduinoOTA.begin(WiFi.localIP(), "MyMKR_wifi_1010", "password", InternalStorage);`

and get the correctly formatted label as seen above.

I will assign this issue to @AlbyIanna to get it fixed and move on.
Thank you @me-no-dev for your patience

kittaakos, 2022-08-18:
> We should resort to displaying the value of `label` coming out of the Discovery.

Unfortunately, the IDE2 does not receive this information from the CLI. Although the [`properties`](https://github.com/arduino/arduino-cli/blob/9fe8560e917ec097b4454f462363e049358017c0/rpc/cc/arduino/cli/commands/v1/port.proto#L32-L33) are available, it does not contain the desired label. The IDE2 cannot do much without CLI changes.

See https://github.com/arduino/arduino-ide/issues/1331#issuecomment-1219491387

---

I update the issue title to reflect that the `label` is the expectation, not the `hostname`.

closed 2022-08-25 by kittaakos