arduino/arduino-ide #583: Allow one sketch to be uploaded to, and serial-monitored on, multiple ports

StephenJD, 2021-10-31:
**Is your feature request related to a problem? Please describe.**
I am testing a sketch on multiple boards (various different Arduino flavours). If I try to open the IDE twice for the same sketch, it simply reverts to the original. To persuade the IDE to give me multiple windows I have to have a differently named sketch for each board. (I cannot find a way to launch multiple IDE's from Explorer.)
This is tedious, because any changes I make to the "master" sketch, I have to copy and paste to all the others. It also means I have multiple sketches on disk, only one of which I actually need.

**Describe the solution you'd like**
The nicest solution would be to enable one IDE to upload to, and monitor, multiple ports, instead of just one. If this is not possible, then when "reopening" a sketch that an IDE is already showing, have it ask if you want to allow multiple IDE's for this sketch. As a last resort, allow windows explorer to launch multiple instances of the IDE.

**Describe alternatives you've considered**
Use Arduino 1.8 which allows windows explorer to launch multiple IDE's with the same sketch.

**Additional context**

per1234, 2021-12-21:
The IDE now does have the capability to open the same sketch in multiple windows via **File > New Window**.

It can be awkward at times though because the most recently opened sketch is loaded (equivalent to the behavior when starting the IDE), which is not necessarily the sketch that was open in the window you accessed **File > New Window** from.

So I'll leave this open for now in case the UX experts want to take a look. I suspect that **File > New Window** was introduced as a side effect of updating to a new version of the Eclipse Theia IDE framework.

StephenJD, 2021-12-22:
I cannot get the behaviour you describe in Arduino 2.0.0-rc3. If I attempt to open the same sketch again, it reverts to the original window.

per1234, 2023-08-27:
> The IDE now does have the capability to open the same sketch in multiple windows via **File > New Window**.

The feature stopped working at 2.0.0-rc7 and was later removed via https://github.com/arduino/arduino-ide/pull/1071