arduino/arduino-ide #585: Problem with #include

jmiszczak83, 2021-10-31:
**Describe the bug**
Cannot include Mouse.h or Keyboard.h

```
#include <Mouse.h>
void setup() {
    Mouse.begin()
}
void loop() {
}
```
Received Error:
`In file included from /home/jmiszczak/Arduino/mouse-test/mouse-test.ino:1:0:
/home/jmiszczak/elektronika/arduino-1.8.16/libraries/Mouse/src/Mouse.h:29:2: warning: #warning "Using legacy HID core (non pluggable)" [-Wcpp]
 #warning "Using legacy HID core (non pluggable)"
  ^~~~~~~
/home/jmiszczak/Arduino/mouse-test/mouse-test.ino: In function 'void setup()':
mouse-test:3:3: error: 'Mouse' not found. Does your sketch include the line '#include <Mouse.h>'?
   Mouse.begin()
   ^~~~~
exit status 1
'Mouse' not found. Does your sketch include the line '#include <Mouse.h>'?
`

**To Reproduce**
Just paste code and verify.
Code works, if i place #include in the same line with object that I want to use:
```
void setup() {
#include<Mouse.h>  Mouse.begin()
}
void loop() {
}
```

**Expected behavior**
Global #include should work

**Desktop (please complete the following information):**
Windows 10 and Ubuntu 20.04

per1234, 2021-10-31:
Hi @jmiszczak83. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there:

https://forum.arduino.cc/

closed 2021-10-31 by per1234

jmiszczak83, 2021-10-31:
@per1234  but this is a bug...