arduino/arduino-ide #586: Upload fails if port is open in Serial Monitor of another window

per1234, 2021-10-31:
## Describe the bug

It is not uncommon for a user to have multiple windows open with the same port selected in each. If the Serial Monitor is open in one of those windows, uploads from the other window will fail.

## To Reproduce

1. Connect any Arduino board to your computer.
1. Select the board and port in the Arduino IDE menus.
1. Open Serial Monitor (e.g., **Tools > Serial Monitor**).
1. Select **File > New** from the Arduino IDE menus.
1. Select **Sketch > Upload** from the Arduino IDE menus.
   Note that the upload has failed with an error something like "`avrdude: ser_open(): can't open device "\\.\COM7": Access is denied.`"

## Expected behavior

Handle sharing port usage between windows automagically.

**OR**

More effectively communicate the cause of the upload failure and the required action to the user.

**OR**

Clearly document the cause of the upload failure and the required action.

I would recommend taking the approach of "so you have x problem, here is the solution ..." rather than "here is how to upload/use Serial Monitor properly". The reason is that the user is much more likely to go searching for a specific solution after encountering the problem than they are to proactively study the documentation in advance of attempting an upload. With this approach, the appropriate location for the documentation would be [the Help Center](https://support.arduino.cc/hc/en-us/sections/360003198300). The [classic Arduino IDE points the user there](https://github.com/arduino/Arduino/blob/1.8.16/arduino-core/src/cc/arduino/packages/Uploader.java#L179) in the event of a failed upload. The article should be titled with the unique components of the various common error messages that occur when attempting to upload to an open port. It should also take into account that there are multiple possible causes of this error.

## Desktop

- OS: Windows 10
- Version: 2.0.0-beta.12-nightly.20211028
Date: 2021-10-28T03:03:25.353Z
CLI Version: 0.19.1 alpha [718bbbf2]

## Additional context

Originally reported at https://github.com/arduino/arduino-ide/issues/271#issue-840803269

### Comparison to classic Arduino IDE

This is not usually a problem in the classic Arduino IDE because all windows share the same Serial Monitor and are able to juggle the control of the port between the upload process and Serial Monitor automagically.

The exception is when the user has opened multiple instances of the Arduino IDE, in which case it behaves the same as Arduino IDE 2.x in that each instance has its own Serial Monitor and can not control the Serial Monitor of the other instance. However, this multiple instance usage is not very common, and more often employed by advanced users who are easily able to identify the cause of the upload failure that results from these conditions.

per1234, 2021-10-31:
I was not able to reproduce this bug with 2.0.0-beta.3, 2.0.0-beta.4, 2.0.0-beta.12, or 2.0.0-beta.12-nightly.20211028. For this reason, I'm not able to determine whether the bug has been resolved since the time of the report.

@ubidefeo I have given you the initial assignment to see whether you are able to reproduce it. Once that is done, I'll will follow up on GitHub either to get a confirmation from the up with the reporter or to mark it resolved, depending on your findings.

ubidefeo, 2021-11-01:
hey @per1234 
I am able to reproduce this, let's bring it into the backlog.
I believe the handling of Serial ports should be done automagically.
If a port is open and an upload towards the same port is initiated, the connection should be closed and upload started.
I also think the previous connection should not be automatically re-started

u.

tigoe, 2022-03-11:
This is still failing. Steps to reproduce:

1. make a new sketch. I used AnalogReadSerial from the Basics examples
2. upload to a board
3. open the serial monitor
4. make a new window
5. make a new sketch. I made a variation on AnalogReadSerial, adding a line that printed "test"
6. Try to upload. You will get this error:
````
Error: Request upload failed with message: Upload error: Failed uploading: uploading error: exit status 1
````

There's no indication that the open serial monitor in the other window is the problem.  This will frustrate the hell out of beginners.

closed 2022-06-08 by fstasi

cnyrr, 2023-07-23:
It is still broken. Can't upload unless serial monitor is closed.
```
avrdude: ser_open(): can't open device "\\.\COM6": Access is denied.
Failed uploading: uploading error: exit status 1
```

bakedPotatoLord, 2025-09-01:
same here. I'm experiencing this on the `2.3.6` version on a windows 11 machine and an Arduino uno.

per1234, 2025-09-01:
Hi @bakedPotatoLord. The bug that caused this problem was fixed three years ago, in version 2.0.0-rc7. So the problem you are experiencing is definitely not the same.

Please post over on Arduino Forum. That will be the appropriate place for us to work with you in investigating the problem you are having:

https://forum.arduino.cc/c/development-tools/ide-2-x/93

I closely monitor the forum so I'll be sure to see your post over there.